christianhaitian/arkos #222: scummvm launch script missing rom variable for retroarch

southoz, 2021-01-29:
The shell script /usr/local/bin/scummvm.sh is missing the rom variable declared in es_system.cfg and could be updated as below to allow directly loading the game.

**Option 1 - Requires going into the subdirectory to click on the gamelist entry. Maintains the option to start into menu**

```
#!/bin/bash

if  [[ $1 == "standalone" ]]; then
   cd /opt/scummvm
   ./scummvm
else
   if  [[ $3 == "/roms/scummvm/menu.scummvm" ]]; then
      /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so
   else
      /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so "$3"
   fi
fi
```
Example - Day of the Tentacle

game folder = /roms/scummvm/tentacle/
scummvm file = /roms/scummvm/tentacle/tentacle.scummvm
file contents = tentacle

gamelist.xml entry
```
  <game>
    <path>./tentacle/tentacle.scummmvm</path>
    <name>Day of the Tentacle</name>
    <desc>Dr. Fred's mutated purple tentacle is out to take over the world. And only you can stop him. Travel through time with three outrageous characters in this wacky graphic adventure which features cartoon-style animation and over 100 zany sound effects.</desc>
    <rating>0.7</rating>
    <releasedate>19930101T000000</releasedate>
    <developer>LucasArts</developer>
    <publisher>LucasArts</publisher>
    <genre>Adventure-Adventure / Point and Click</genre>
    <players>1</players>
    <image>./images/tentacle.png</image>
    <thumbnail>./thumbnails/tentacle.png</thumbnail>
    <video>./videos/tentacle.mp4</video>
    <genreid>514</genreid>
  </game>
```
Add the game through the scummvm menu once and then start from the subdirectory object

**Option 2 - Tricks EmulationStation into a flat folder structure to select the gamelist entry from the root scummvm folder. Maintains the menu option to start into menu**

```
#!/bin/bash

if  [[ $1 == "standalone" ]]; then
   cd /opt/scummvm
   ./scummvm
else
   if  [[ $3 == "/roms/scummvm/menu.scummvm" ]]; then
      /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so
   else
      rom=`echo $3 | grep -oP "\/(?:.(?!\/))+$"`
      rom="${3}${rom}"
      /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so "$rom"
   fi
fi
```
game folder = /roms/scummvm/tentacle.scummvm/
scummvm file = /roms/scummvm/tentacle.scummvm/tentacle.scummvm
file contents = tentacle

Add the game through the scummvm menu, exit and then start from gamelist entry.

gamelist.xml entry
```
  <game>
    <path>./tentacle.scummmvm</path>
    <name>Day of the Tentacle</name>
    <desc>Dr. Fred's mutated purple tentacle is out to take over the world. And only you can stop him. Travel through time with three outrageous characters in this wacky graphic adventure which features cartoon-style animation and over 100 zany sound effects.</desc>
    <rating>0.7</rating>
    <releasedate>19930101T000000</releasedate>
    <developer>LucasArts</developer>
    <publisher>LucasArts</publisher>
    <genre>Adventure-Adventure / Point and Click</genre>
    <players>1</players>
    <image>./images/tentacle.png</image>
    <thumbnail>./thumbnails/tentacle.png</thumbnail>
    <video>./videos/tentacle.mp4</video>
    <genreid>514</genreid>
  </game>
```

christianhaitian, 2021-01-29:
Cool.  I wasn’t focused on lr scummvm as most seem to be happier with the standalone scummvm.  I will add one of these fixes in the next update. Thanks for sharing.

christianhaitian, 2021-01-30:
Option 1 has been incorporated in this evening's update.

closed 2021-01-30 by christianhaitian