christianhaitian/arkos #223: Open games that scraper does not recognize

VorgonBattlecruiser, 2021-01-29:
I use a lot of romhacks, fantranslations etc.
Scrapers dont recognize those or label them incorrectly.

There seems to be no way to just use the regular Retroarch File explorer, it always boots to some kind of simplified UI.

christianhaitian, 2021-01-29:
Can’t do anything about the rom hacks not scraping.  Their custom hacks.

File explorer within retroarch works fine.  You’re probably wanting the ps3 xmb style?  If so, follow this guide from retro game corps on setting this up. https://retrogamecorps.com/2020/11/15/arkos-rg351p-guide/
Check the notes area.

closed 2021-01-29 by christianhaitian

VorgonBattlecruiser, 2021-01-29:
Is there a way to start retroarch directly without having to chose a rom in the simplified UI first?
I actually prefer ozone as a theme, but i dont understand what that has to do with anything.

christianhaitian, 2021-01-29:
Oh sorry, I must have misunderstood your question.  It is possible but you'd have to ssh into the unit and make some changes.:

sudo systemctl enable ssh
sudo systemctl disable emulationstation
sudo systemctl enable  rc.local

edit rc.local and enter the following:
/usr/local/bin/retroarch

Then reboot.

VorgonBattlecruiser, 2021-01-29:
Please disregard any prior statements. Everything works flawlessly now.
I think something went wrong during the initial flash with win32diskimager.

UI was unresponsive and glitchy and i could barely do anything.

Now everything runs well.
Stellar effort! Thank you so much for all your hard work!