christianhaitian/arkos #224: Save State Issue

ZillaTBO, 2021-01-30:
I am using the RG351P with ArkOS v 1.5 (01282021)

so here is the issue, while using the Playstation emulator i can make save states, however there is no way to load them.  ive set up all my hot keys to to save and load the save states. the Save hotkey works, shows the saving bar. scrolls to 100% and gives the check mark. the load hotkey does nothing. then i tried to do manual save states. going into retro arch selecting save, it runs a save, even makes a save state file in my game directory, however the load save state icon is not there.

troubleshooting steps i have taken so far.  
1. going into the file directory and deleting the old save states. no resolution.
2. taking the SD card and loading it on the my pc to see if there was an error that needed repair via windows. no prompt for repair no obvious issue found.
3. switched the core to run on the Retro arch 64 version.  it worked briefly but after closing and reopening the rom the same issues with the 32 version have started.

save states seem to have worked normally on each version of retro arch once, then on future boot ups this issue happens.

to help you out. the game i was playing when i came across this issue was Final Fantasy Tactics. also with a secondary Saving issue.
since save states have not been working for me ive been using the in game saves.  during the first 10 hours of game play with the game, i could save in game as normal. and then when i was done playing shut the core down using the Select + Start Start hot key. the saves would be fine. however after a slightly longer play session, about 2 and a half hours, i saved, closed the core using those hot keys and all the saves and progress had been reverted to the state prior to that game session. losing a ton of progress and time.  frustrating but i have at least found a work around for this issue for now.
going into retro arch and closing the content first, then closing retro arch seems to resolve the issue,  an extra step but possibly something else you may want to look into.

sorry for the long post but i wanted to give you as much info on the issue as ive found to help you with troubleshooting this issue. 
also. the save state issue is an issue my brother is dealing with as well. he is also on an RG351p. also the same version of ArkOS.

### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-01-30:
This issue is very odd.  First, if the issue is happening between retroarch 64 and 32, then this means it’s happening between 2 different emulator cores as the retroarch 64 PSX emulator is duckstation while on the reteoarch 32 the emulator is pcsx_rearmed.  It’s very strange that the issue happens fairly consistently the same between 2 completely different emulators.  So My thinking is that there’s either an issue both retroarch engines or an issue with your hotkey changes. By default, the image is set to use hotkey+r1 for savestates and hotkey+l1 to load savestates.  You indicated you changed this.  Are you sure you’re not changing your savestate slots by accident between sessions?  Or did you enable Increment Save State Index Automatically in the retroarch menu?  If that’s not the issue, are you experiencing this with any other system emulators?  I tried the following scenarios and can’t replicate this issue yet using Ape Escape

launch ape escape, savestate once started in new game.  Load savestate with no issues using default hotkeys,

Exited retroarch via hotkeys.  Load ape escape again, loadEd savestate without issue using hotkeys.

Shutdown the unit.  Boot the unit.  Launch ape escape, load savestate, no issue.  Saved over previous savestate with no issue.  Loaded new savestate,  no issue.

ZillaTBO, 2021-01-30:
I’ll do some more testing and I’ll even take a video to show you what’s going on if that helps.  Also, those are the same hot keys I use for save state and load state. Didn’t realize that’s what they were already set to.   Would you like video of the issue so that you can see what’s going on?

christianhaitian, 2021-01-30:
How about you grab another sd card, and image arkos onto it, update it, then load your game on it, then try that without making any hotkey changes there and see what happens.

ZillaTBO, 2021-01-30:
Sounds good. I’ll give that a shot and see what happens.

ZillaTBO, 2021-01-30:
okay, so i flashed a new SD card. and everything is working as it should.  however i took a look at my other SD card (the one with the issue)  i dont think changing the hotkeys, or really remapping them to the same buttons as before is the issue,  when you go into the retro arch menu while the game is running the Load State option is missing. its there on the fresh flash of the OS.

so what im going to do is move all my games and saves off the old sd card and re flash it with a fresh install and go from there.  if the problem comes up again ill be sure to reach out to you again if i can replicate the issue.

thank you for your time. and keep up the good work on this OS.  its awesome!

closed 2021-01-30 by christianhaitian