christianhaitian/arkos #225: DuckStation doesn't seem to work

MetroidFusion, 2021-01-30:
Hi

When choosing DuckStation for playing PSX games, the screen goes black and instantly exits back to EmulationStation. I tried installing the DuckStation core in Retroarch and load it from there, but it simply won't start any games. Don't know if anything can be done about it, but thought I'd let you know.

Thanks!

christianhaitian, 2021-01-30:
I just tried ape Ape Escape on my end and it loaded load just fine in duckstation. Do you have a proper psx bios loaded in your bios folder?  Also, what extension are your psx games?  Could be duckstation doesn’t support a particular extension that pcsx_rearmed does.

dredd666, 2021-01-30:
Same issue for me. It used to work, but stopped when i tweaked some settings on options. Is there a way to reset the Core options to default?

christianhaitian, 2021-01-30:
Not directly on the device.  I found that going into the duckstation core options and setting the gpu renderer to vulkan causes the emulator to no longer launch.  What has to be done is to manually modify the file located at:

/home/ark/.config/retroarch/retroarch-core-options.cfg
and set
 duckstation_GPU.Renderer = "Software"

You can access this location via ssh using a program like winscp or putty or through the web based arkos filebrowser by doing the following:

with your wifi adapter plugged in and connected to your wifi signal, enable remote access from the options section

1. From a computer, tablet, or smarthphone go to http://rg351p
2. Go to settings, user management, then hit the little pencil at the end of the ark username line.
3. In the scope box, replace /roms with /
4. scroll down and hit save.
5. Then click the my files button the top left of the screen then navigate to the file location above
6. Then make the change in the file and save it.

The other option would be to use a linux computer or vm and make the change there.  In a future update, I'll include an option to reset the retroarch and retroarch32 core option files to the default options included with ArkOS.

Thanks,

MetroidFusion, 2021-01-30:
> I just tried ape Ape Escape on my end and it loaded load just fine in duckstation. Do you have a proper psx bios loaded in your bios folder? Also, what extension are your psx games? Could be duckstation doesn’t support a particular extension that pcsx_rearmed does.

I just researched which file formats DuckStation supports, and unfortunately it does not support PBP files. So your assumption was right! Thanks for clearing it up :)

christianhaitian, 2021-01-30:
Awesome.  Enjoy!

closed 2021-01-30 by christianhaitian