christianhaitian/arkos #230: Screen cut off MD / Sega CD & Core Override Issue

leighg101, 2021-01-31:
Console: RG351P
ArkOS: latest version

Hey there, here are some bugs I’ve found.

BUG 1:
I’ve noticed that sometimes when you load a game on the MD / Sega CD the screen is cut off and you need to press start + select to reset it to normal

video here I showed to Retro Game Corps:
https://twitter.com/eufeion/status/1355817386457358336?s=21

BUG 2:
When I load a game (mega drive / Sega CD) I find that RGA is alway turned off even though I’ve saved to content directory override after turning it on. I did save accidentally prior to Core Override instead of directory before I did directory override but after saving the settings to core directory, the main core override loads instead of the directory one

christianhaitian, 2021-01-31:
Interesting. So I can replicate this, what are the retroarch settings you have changed?

christianhaitian, 2021-01-31:
Also, for content override, I just tested saving a core override with 4:3 and rga scaling off, then a content override with 4:3 and RGA scaling on and RGA scaling was on by default.  I even tested saving RGA scaling off again and saving core override and still RGA scaling loaded as contetnt overrides override core overrides as it should.  Please check retroarch settings then configuration and verify that Load Content-Specific Core Options Automatically is set to on.

leighg101, 2021-01-31:
I’ve checked and it only happens on Mega Drive / Sega CD. The only settings I’ve changed are RGA & Scaling. The scaling stays it’s just the RGA. I checked the other settings and the auto load is on. I’ve done a video below (muted) an it shows the problem and also the screen normal then on a second load it being cut off with me having to press start + select to rectify it before going into the Retroarch menu 

https://youtu.be/H8NFF-sL7Ww

christianhaitian, 2021-01-31:
I finally replicated the screen cutoff issue by setting Aspect Ratio to Core provided.  Using 4:3, the issue doesn’t occur for me.  I suspect there’s an inconsistency issue with the genesis core causing that.  I’ll see if there’s an updated core available that maybe fixes that issue.  As for the RGA scaling, override works without issue on my end.  Please post a pic of your settings>configuration section.

leighg101, 2021-01-31:
[image]

hope that helps

christianhaitian, 2021-01-31:
The jpeg is not loading.

leighg101, 2021-01-31:
Try this 
https://imgur.com/a/qCJvUDL

christianhaitian, 2021-01-31:
That worked.  Why the last option (Use Global core options file) off?  Try turning that on and save it and see if there's a difference.

leighg101, 2021-01-31:
I get this error when trying to save, it’s okay on the Mega Drive & Sega CD I get this 

https://imgur.com/a/jWSoc0J

christianhaitian, 2021-01-31:
Using file manager from the options section, go to the /home/ark/.config/retroarch/config folder and delete the Genesis Plus GX folder.  Then try again.

leighg101, 2021-01-31:
Deleting that folder seems to have fixed the RGA, Screen cut off & the settings issue.

tested with multiple games on both the Mega Drive & Sega CD as well as powering off an testing again with other games and all seems ok now

christianhaitian, 2021-01-31:
Oh nice.  Enjoy!

closed 2021-01-31 by christianhaitian