christianhaitian/arkos #231: Low Battery Warning configurable?

NickDevoctomy, 2021-01-31:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

no

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

yes, there is a battery level related bug but not this specific issue

### Describe the Issue (If applicable)

When battery got below 10% it refused to go into the menu, and instead displays a warning with the vibration motor permanently on.  Is this intentional?  It's similar to what it does on a normal boot except that the vibration just keeps on going.

### How can the issue be reproduced? (If applicable)

Let battery fall to around < 10%

### What device are you using?

Anbernic RG351p

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

1.5

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-01-31:
That's an odd issue.  I haven't experienced this issue myself and haven't heard this being an issue for anyone else.  At least not yet.  I suspect the unit simply did not have enough power left and was starting to have issues properly powering the device and started having issues.  Has this happened more than once?

spot1701, 2021-02-01:
I can say that I've also had this happen. But it only happened once, and I didn't equate it to lack of battery. I will try and replicate.

christianhaitian, 2021-02-01:
So while the unit was on, as it continued to fall to below 10 percent is when the issue occurred?  Or were you attempting to turn the unit on at less than 10 percent and this issue occurred?

NickDevoctomy, 2021-02-01:
Hi there,

Sorry for the delayed reply, basically it was running ok.  I turned the device off to add some more ROMs to the SD card, put it back in, booted it up, and it's at that point it started telling me the battery was too low.  When I turned the device off I believe it would have been on, or just above 10%.

I plugged the device into my PC USB, and turned it on, and it booted up fine, just in case there was any weird SD card related issues.

christianhaitian, 2021-02-01:
Ok.  That's a known occurrence with the rg351p and other rk3326 devices.  If you know the battery was definitely low, like under 30 percent, the unit can have issues powering on since the battery capacity, if low, may not complete the initialization process correctly.  If you're aware that the unit is below 50% capacity, charge the unit asap to avoid strange issues like this.

closed 2021-02-01 by christianhaitian

NickDevoctomy, 2021-02-01:
Great, thanks for the info, much appreciated.