christianhaitian/arkos #233: support for odroid go super

jdalmanza, 2021-02-01:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
Image does not boot on Odroid Go Super

### How can the issue be reproduced? (If applicable)
Flashing image to new SD card, or using SD card from working Odroid Go Advance system

### What device are you using?
Odroid Go Super

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

1.5 downloadable image provided (base)

### Anything else you'd like to include that may help to review this issue or feature request?
Hard Kernel forum introduces changes to uboot, boot.ini and kernel for Odroid Go Super. 
https://forum.odroid.com/viewtopic.php?t=41283

christianhaitian, 2021-02-01:
I have no plans to support the Odroid Go Super with ArkOS at this time.

closed 2021-02-01 by christianhaitian

SuperBadger77, 2021-03-04:
Any plans in the future to support OGS? Love the OS - would be great to see on the OGS :-)

christianhaitian, 2021-03-04:
No guarantees but it may work with the next online update when that releases.  It would be with the RGB10 version if it does work.