christianhaitian/arkos #236: Odroid Super Support

therustyrobot, 2021-02-03:
The Odroid Go Super is now being delivered and several of us were wondering if you would be able to support an update for that device. The hardware is supposedly the same apart from the screen.

https://www.reddit.com/r/SBCGaming/comments/l5thu6/does_arkos_run_on_odroid_go_super/

christianhaitian, 2021-02-03:
I have no plans to support the Odroid Go Super with ArkOS at this time.

closed 2021-02-03 by christianhaitian

jdalmanza, 2021-03-20:
just FYI, with your 1.6 update from 18-MAR it does support odroid go super. I updated my oga image to this update as it was available, put the sd card in my ogs ... viola! so, conceivably if you wanted to post an updated disk image with this update it will boot on an odroid go super fresh. 

v1.6 (03/18/2021)
...
- Update uboot and bootini to latest version <-- this was all that it took ;-)