christianhaitian/arkos #238: Emulator no longer opens post-CHD conversion.

VorgonBattlecruiser, 2021-02-06:
I applied the unworked romhack to Alundra 1 PSX bin/cue files (checked hash on my dump beforehand) and the game ran really well on my RG351.

Afterwards i converted all my games to CHD with the latest CHDman. (I made sure to use the CUE files for conversion).
The conversion displayed no errors.

I select the image, it goes to a black screen and returns to the game selection, other converted PSX games work fine.

Is this a compatibility issue? I tried with Duckstation and pcsx rearmed in both retroarch and retroarch32.

Firmware and retroarch cores are all on the latest versions.

christianhaitian, 2021-02-06:
It sounds like the game simply won't work as a CHD file.  You may just need to leave it as a cue/bin file.

VorgonBattlecruiser, 2021-02-06:
I just couldnt understand why this game out of all my games just wont work.
Still dont, to be honest, but i got it to work!
I tested PCSX rearmed with rumble as a core and duckstation.
Neither worked for me.
Out of sheer hopelessness i also tested pcsx rearmed without rumble... and that worked!

I honestly thought the only difference between those was the rumble-support.

Do you have any idea why this is happening?

christianhaitian, 2021-02-06:
The rumble based version is more recent.  Could be something in the updated code.  Enjoy!

closed 2021-02-06 by christianhaitian