christianhaitian/arkos #239: ScummVM Virtual Keyboard

mudHOAX, 2021-02-06:
ScummVM should be built enabling options for the virtual keyboard
https://wiki.scummvm.org/index.php/Virtual_Keyboard#:~:text=By%20default%20the%20virtual%20keyboard,when%20running%20the%20configure%20script.

At present, it is impossible to progress with certain games, such as Leisure Suit Larry 7 without connecting an external keyboard.
It would be great if we could have native support for the virtual keyboard.

christianhaitian, 2021-02-06:
Thanks.  I'll recompile and include in the next update

christianhaitian, 2021-02-07:
Hello,

I finally had a chance to recompile this emulator with the virtual keyboard enabled.  Can you help me test this?  Please follow the instructions below to test this and provide feedback on this when you can including any suggested key remaps if any.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

1. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

1. Once completed. Reboot the device and test the new scummvm standalone for virtual keyboard and ensure nothing else has been negatively impacted, performance wise. Let me know your results and if good, I will include it as part of the next updated.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/5939907/arkosbackup.tar.gz)

christianhaitian, 2021-02-09:
Hi, any progress or plan to test this before release?  The virtual keyboard seems to be there on my end with this new build.
Thanks

mudHOAX, 2021-02-09:
Hi, sorry for the delay, I missed the email notifications. 

I've tested this and after remapping the buttons to display the virtual keyboard it appears to be working as expected.
I've not got a bunch of games to test against, but Leisure Suit Larry 7 accepts the keyboard inputs (note the game thread is paused whilst the keyboard is open, but this is expected from the ScummVM documentation).

I've also not noticed any negative performance issues in either LSL7 or Secret of Monkey Island.

Huge props on this and the quick turn around!

Thanks.

christianhaitian, 2021-02-09:
Awesome!  Thanks for confirming!

closed 2021-02-09 by christianhaitian

SaintHarris, 2021-03-17:
Hi, what is the default button on the RG351P to open the virtual keyboard? 
Thanks.