christianhaitian/arkos #241: TI99/4A Parsec game controls very slow

Mika73, 2021-02-09:
Just updated latest ArkOS. TI99/4A emulator Parsec game controls are very slow. First ship move everything ok but when enemy ships come it starts move extremely slow. Parsec game is maybe most famous TI99 game.

christianhaitian, 2021-02-09:
It's how the game is designed.  It's inertia based and you have to pick your speeds based on how you want to control it using 1(B button), 2(X button), or 3(Y button).  See this [video](https://youtu.be/giOgvJ0n9cs?t=49) for more information about this.

closed 2021-02-09 by christianhaitian

Mika73, 2021-02-10:
Thanks. I didn’t know that you and can choose speed even played this with TI99/4A many times. I guess I had full speed when start game with joystic fire button.