christianhaitian/arkos #242: psx n64 dreamcast is very slow. Especially the slow psx is a big problem. Hopefully this has been fixed in order to continue using Arcos. Please.

haveanair1, 2021-02-09:
Currently using version 210203. psx n64 dreamcast is very slow. retroarch32 pcsx_rearmed_libretro.so Especially the slow psx is a big problem. PSX is slower than rg350. Hopefully this has been fixed in order to continue using Arcos. Please. n64 rice core is The speed is fine, but the graphics break a lot. If you know the option value that can drive PSX n64 Dreamcast better, I would like to check it out together.

christianhaitian, 2021-02-09:
First i'm hearing of slowness with psx.  Have you changed any default settings in retroarch like threaded video?  Also, are you using a bios for psx and dreamcast?  My psx games run around 58fps or better.  Example games being ape escape and
 Mr. Driller.  For Dreamcast, performance can vary but i get 50fps or better with power stone.

haveanair1, 2021-02-10:
The answer was helpful. Solved it. Thank you.

closed 2021-02-10 by christianhaitian

haveanair1, 2021-02-10:
I set 128 gigabytes. It was confirmed that DC and AW NAOMI slowed down as the game was put in a certain amount and the retro arc setting value increased. Many of the games of AW and NAOMI crash and out.