christianhaitian/arkos #246: ArkOS new strange behaviours (e.g. roms not appearing, cannot add favourites)

markusveralius, 2021-02-15:
I have been using ArkOS problem free for many weeks now. In the last few days, I have noticed some strangeness with the OS, including:

- Roms added to the SD card failing to appear
- Favourites added disappear from the favs list once I restart the system
- Last Played no longer updates with recently played games
- I delete roms from within ArkOS but they reappear once I restart the device.
- Some roms now appear twice.

Do you have any suggestions as to what the problem may be?

Thank you.

christianhaitian, 2021-02-15:
From the emulationstation menu (start button), go to advanced settings, parse gameslists only, and make sure that setting is set to off.

christianhaitian, 2021-02-22:
Any update on if the above resolved your issue?

markusveralius, 2021-02-22:
Hi Christian,

Thank you for chasing this up! It did fix the issue that new roms loaded
onto the SD card were not being recognised. I can also now delete games.

However, I still have the same problem that I can no longer add to the
favourites list.

On another note, there are some random oddities I have noticed with the OS.
When plugging in a charger, the battery charge level jumps to a new value
immediately (not the same number shown when on battery power).
Also, on occasion the cursor will move several steps up or down by itself
when in emuelec menus.

Thanks for the continued support,
Mark

markusveralius, 2021-02-22:
A final bug I have encountered several times when attempting to update
ArkOS: The device sometimes hangs on the 'Once you proceed...' warning page
before I can click 'ok'.

christianhaitian, 2021-02-22:
The power issue is a known issue due to the kernel.  See [here](https://github.com/christianhaitian/arkos/wiki/Known-issues) for more information on why that happens.  The hanging issue will happen if you keep the unit plugged into power and using wifi on the 351p.

closed 2021-02-22 by christianhaitian