christianhaitian/arkos #247: Half-Life Camera Drift Reocurring.

uv54633, 2021-02-15:
I am on the most recent version of Ark OS from 2/13/2020 and am experiencing the camera in Half-Life moving to the right side slowly. When I pause the game the camera has moved when I paused the game.

christianhaitian, 2021-02-15:
It sounds like your analog controls have more drift.  You can try increasing the deadzone for the following lines in /roms/ports/Half-Life/valve/config.cfg.  You can do this from a Windows, Mac or Linux PC directly if need be.

joy_forward_deadzone "10000" -> joy_forward_deadzone "20000"
joy_pitch_deadzone "10000" -> joy_pitch_deadzone "20000"
joy_side_deadzone "10000" -> joy_side_deadzone "20000"
joy_yaw_deadzone "10000" -> joy_yaw_deadzone "20000"

uv54633, 2021-02-15:
Ok, thank you, I thought it was possible that the dead zone issue reoccurred as there is no problem in other games. I just didn’t know how to adjust the deadzone.

closed 2021-02-15 by christianhaitian