christianhaitian/arkos #251: Request for RetroArch ZX81 core to be implemented

deadkat37, 2021-02-16:
Hi, would it be possible to implement the RetroArch ZX81 core in ArkOS?

The ZX81 was a primitive black & white personal computer released in 1981 which kick-started the computer boom in the UK and was followed by the more well known and successful ZX Spectrum.

I don't know how widely known the computer is outside of the UK, but it would mean a lot to me to be able to play much loved games like 3D Monster Maze, 3D Defender, Mazogs and Mikro-gen Scramble on the RG351P.

Many thanks for all your hard work on ArkOS.

christianhaitian, 2021-02-17:
I’ll review for consideration.

christianhaitian, 2021-02-20:
This has been accommodated with today's update.  See [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#zx-81) for rom loading information.  Enjoy!

closed 2021-02-20 by christianhaitian

deadkat37, 2021-02-20:
Thank you so much for implementing this :)

Do you have a patreon account (or nominated charity) where I can show my appreciation?

christianhaitian, 2021-02-20:
None at this time.  Enjoy!