christianhaitian/arkos #252: [Bug/Update Issue]

binglywhopper, 2021-02-17:
RG351p
Firnware: ArkOS (the one with the TI-99/4a update)
Clock HI

Issue:
When trying to update I get:

There was an error with attempting this update

I tried turning on error logs in advanced settings(?) But only found /home/ark/esupdate.log that seems older but has the same error message.  Any ideas?  Love your work!

christianhaitian, 2021-02-17:
Means that it couldn’t successfully complete the download.  Check your wifi connection and try again.

binglywhopper, 2021-02-17:
Dude!  It's you!  Awesome!
I tested for gateway on several other devices on my hime network.  All working.  Deleted and rescanned/connected to my router.  Network info shows per normal (192.168.1.24...   gateway 192.168.1.1.)
Got the same error when trying to update.
I've tried scraping (from the metadata menus) and get zero results for popular games like pac man, mario etc.  So yeah, definitely a network issue.  The only anectdote I have with this is that I randomly did my daily check for updates and got, "no new updates etc.."  a fre minutes later I saw the latest update with the fixes for global macros so I went to grab it and got that error.  I know this is more like stuff that would happen back in the 80s with phone disconnects, but do you think i tried to update at exactly the "wrong time"?  Thanks so much again :)

christianhaitian, 2021-02-17:
Based on what you’re sharing, you may have a bad adapter.

christianhaitian, 2021-02-17:
What is your network adapter?

binglywhopper, 2021-02-17:
I have the specific combunatiin as suggested in the retrogamecorps video,  
*update*  just tried updating again, got, (No more updates available, check back later"  but when I rebooted the splash screen said  02132021-1.

christianhaitian, 2021-02-17:
So you are up to date now

christianhaitian, 2021-02-17:
The last update of 02132021-2 for the global hotkeys was for the rgb10, rk2020, and the OGA only.  Not for the rg351p.

closed 2021-02-17 by christianhaitian

binglywhopper, 2021-02-18:
You're the best.  Thanks for walking me through all that :)