christianhaitian/arkos #253: NDS Games are not running and Emulator is not shown

diemorgensterne, 2021-02-17:
Hello Christian 

I run into the issue that NDS games are not starting. I have tried to uninstall and reinstall the core, have done restarts etc. What looks strange to me is, that the NDS Emulator can not be found in the options at all. 

Thanks in advance for having a look into it
Regards from Germany
Tommy

christianhaitian, 2021-02-17:
Drastic is the sole emulator available in ArkOS for DS.  You won’t see an option for it because Drastic is the only option for DS games.  It is not a Retroarch core.  Make sure your NDS games are in .zip or .nds format and are valid.  If in .zip format, make sure the zip doesn’t contain any other files but a .nds rom.

diemorgensterne, 2021-02-17:
The rom are purely .nds files and run smooth on the NDSL with R4 Card. When I start a game boots into the LogoScreen and flickers and then back to the OS again :( 

Thanks for your help... is there a way to support the work you have done. For excample thru paypal or so?

Take care, Tommy

christianhaitian, 2021-02-17:
Have you been making any changes to the system at all?  Such as via ssh?

diemorgensterne, 2021-02-17:
yes, I tried once to transfer files thru ssh but haven't really made any change in the system itself - is it somehow possible to set the system back to the original status without touching the games partition?

diemorgensterne, 2021-02-17:
I added also some bezel based on a guide by retrogamecorps

christianhaitian, 2021-02-17:
What all folders do you have within the roms/nds folder?

diemorgensterne, 2021-02-17:
images & video folders from screen scraper only

diemorgensterne, 2021-02-17:
how dump I am, sorry and thanks for your help on this - the last thing I need to get working is Sega CD if you have not a direct hint what I need to do here I going to search the web first - this issue can be closed, again thanks for your help and let me know how I can support you

christianhaitian, 2021-02-17:
For Sega CD, make sure you follow the bios and rom file requirements [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#sega-cd).  If you need help with verifying you have the right bios, you can check the md5 checksum of the files against the ones referenced [here](https://retropie.org.uk/docs/Sega-CD/).  Good luck and have fun!

closed 2021-02-17 by christianhaitian