christianhaitian/arkos #255: Mupen64 core dont working

X3meOS, 2021-02-17:
Hello :)

i am using the newest version of ArKOS on my purple 351p.
But there is one single problem i cant solve.

in the emulation settings of n64 i can find Mupen as Standalone.. only as Core. maybe this is normal.
So not in Retroarch32 .. only in Retroarch.
If i wanna start a game now (with mupen selected as core) no game will start.

i tryd every possible setting.

someone has any ideas or expiriences?
on videos i only see people who playing very well with mupen.. but in my case it will not work - i am very sad :D 

Thanks for every hint!

greetings

christianhaitian, 2021-02-17:
This is normal.  Standalone = Mupen64-rice
Standalone may not be able to open the N64 rom archive.  Try just opening a rom that ends in .n64 or .z64

X3meOS, 2021-02-17:
Hallo and thanks for fast Response.
I did this and still get a crash after start a game with mupen :(

christianhaitian, 2021-02-17:
Have you accessed ArkOS via SSH recently to make any changes?

X3meOS, 2021-02-17:
Nope .. maybe a Problem in the bios folder? I copied the bios folder from
Stock Rom to Arkos

christianhaitian, 2021-02-18:
Do the games work with the retroarch n64 emulators?

X3meOS, 2021-02-18:
Yes.. only mupen crash Instantly

christianhaitian, 2021-02-18:
Have you updated ArkOS via Wifi since you installed ArkOS on the sd card?

X3meOS, 2021-02-18:
Yes.. but Not tryd mupen before the sync

christianhaitian, 2021-02-24:
What kind of computer are you using to load your games onto the SD card? Is it a Mac or PC?  Which?

christianhaitian, 2021-02-26:
Any update to my last question?

X3meOS, 2021-02-26:
Hello,

Win10 64bit

christianhaitian, 2021-02-26:
Ok.  Try a fresh image as I have no other suggestions at this point.

closed 2021-02-26 by christianhaitian