christianhaitian/arkos #256: RG351P stuck on ArkOS partition alteration.

JBR0808, 2021-02-18:
[image]

After installing the ArkOS image and putting the new micro sd card into my RG351P, it’s been stuck at “The partition has been altered.” For about 15 minutes (Image included above, hopefully). Did something wrong happen? Or do I have to let it sit for a while longer just incase it’s setting everything up still?

Info that might help:
RG351P
128GB Lexar Professional Micro SD

christianhaitian, 2021-02-18:
There's been reports that Lexar micro SD cards don't seem to work with ArkOS.  Make sure you didn't use Balena Etcher to image this card.  If you did, try with either USB IT or win32diskimager instead.  If that doesn't work, try either an authentic Sandisk or Samsung uSD card if you can.  If not, give 351Elec a try.

JBR0808, 2021-02-18:
> There's been reports that Lexar micro SD cards don't seem to work with ArkOS. Make sure you didn't use Balena Etcher to image this card. If you did, try with either USB IT or win32diskimager [...]

Is there a way I can clear everything from the Lexar and try 351Elec? I don’t have enough time to go out and buy a new one.

christianhaitian, 2021-02-18:
You should just be able to image 351elec over the existing install without issue.

closed 2021-02-18 by christianhaitian