christianhaitian/arkos #257: Saving State causes Retroarch to crash and re-load this also resets the memory card data

MorhionG, 2021-02-18:
This is a random issues and occurs once out of every 7 or 8 savestates. When attempting to savestate using start+X in the retroarch menu, it will cause the game to freeze for a moment and then crash back to Retroarch. This has a 50/50 chance of corrupting the save. Also it will reset any memory card data back to the previous save. The only way to ensure memory card data persists is to save the game, close the game and then go back in.

This happens often enough it's easy to replicate.

christianhaitian, 2021-02-18:
What emulator and games have you had this happen in?

christianhaitian, 2021-02-18:
Also, why are you saving state from the retroarch menu and not using the preset hotkey of Select+R1?

MorhionG, 2021-02-18:
It does it with both methods. After about 8-10 save states it will randomly
crash and reset itself when using the hotkey method as well. So I started
using the other, thinking it may have just been a bug.

MorhionG, 2021-02-18:
Oh also PSX emulator, updated retroarch core and Saga Frontier

christianhaitian, 2021-02-18:
Try the other pcsx-rearmed core and see if you have the issue there.

christianhaitian, 2021-02-20:
Any update with your testing?

MorhionG, 2021-02-20:
The different core offered no change, still having issues with it crashing
every 10-12 save states.Also it seems to screw up memory card data when
that happens too and loses any 'recent' memory card data and reverts to
memory card date before the save.

christianhaitian, 2021-02-20:
Have you enabled runahead or fast forwarding in your the retroarch settings?  If so, try disabling that and see if the issue goes away.  

https://github.com/libretro/pcsx_rearmed/issues/431

christianhaitian, 2021-02-22:
Just seeing if you’ve tried the suggestion above yet.  Thanks

MorhionG, 2021-02-22:
No frame skipping is enabled or anything like that the only global option
set is bilinear filtering

christianhaitian, 2021-02-22:
Have you tried to see if this issue happens on the original sd card or another OS like 351Elec?

christianhaitian, 2021-02-24:
Have you tried what I suggested above?  Any update on this?

christianhaitian, 2021-02-24:
Also, did you make sure to load a compatible bios per the wiki? https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#playstation-1-psx

MorhionG, 2021-02-25:
Yes, I did.

trying elec sometime this week I really like ArkOS so I didn't want to
reflash honestly

christianhaitian, 2021-02-28:
Are you sure you’re not using auto savestates?

MorhionG, 2021-02-28:
Positive, I have to manually save state each time

iCheetle, 2021-03-03:
same here but when i try to flash it it wont let me

christianhaitian, 2021-03-03:
Unfortunately, I have no other ideas to try on this.  Maybe try the latest version of 351elec or retroarena and see if those are more stable for you.

closed 2021-03-03 by christianhaitian

MorhionG, 2021-03-03:
Nah I'm going to stick with ArkOS I really like it. I'll just keep updating
and see if the bug fixes. It's not GAME BREAKING just annoying lol.

Thanks!

GhostCrab, 2021-05-11:
Sorry to necro a closed issue, but I think the part of this user's issue with his memory card reverting is related to SRAM Autosave Interval being disabled by default. This setting may have been disabled on purpose in the interest of preserving SD cards, in which case I wont rehash an argument that has been made to death in this thread: (https://github.com/libretro/RetroArch/issues/4901)
I would like to point out that while leaving SRAM saves for when RA successfully shuts down makes sense on a desktop system that might close RA periodically, these handhelds that are built to go to sleep instead of shut down could possibly go dozens of in-game hours on some long RPG without switching games and have the console run out of power or crash wiping all progress unless the user was using save-states.
I do like using save states, but in my case I was playing Vagrant Story (PSX) which uses an item stash system where excess items are stashed in some kind of secondary save on the memory card and when my system crashed, the memory card reverted a few hours back but my save states were current and got out of sync, corrupting the item stash. I know it's a corner case and maybe not worth the change to the default configuration, but I wanted to share.