christianhaitian/arkos #258: New ports support for Sonic 1, 2 and CD (like emuelec)?

skullmonkey911, 2021-02-19:
Hi Christian,

have you thought about support for the Sonic Games in the ports (as in EmuElec 4 introduced) in the future? It's a cool feature and the games run perfect on my ODS, would love to have it on my 351m as well.

All the best,
Stephan

skullmonkey911, 2021-02-19:
Here are some further information. 

decompilation ports
[17:15]
https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec-ports/sonic2013/package.mk
[17:15]
https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec-ports/rigelengine/package.mk
[17:15]
https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec-ports/soniccd/package.mk

christianhaitian, 2021-02-19:
I will not be adding anymore ports or modifying existing ports.  See [here](https://github.com/christianhaitian/arkos/wiki/Want-more-Ports%3F) for more information on why and alternative solution not supported by me.

closed 2021-02-19 by christianhaitian