christianhaitian/arkos #259: Sporadic RG351M crashes / battery issues whilst using ArkOS - Advice?

SegaMad, 2021-02-19:
Hi,

I recently purchased an RG351M.

At first, my 351M was working perfectly and I was loving playing the games on it.

Following a couple of days, however, the included card would boot up into ArkOS but sometimes “freeze” on the loading Emulationstation progress bar screen, then sometimes would fully load and play games fine. One time I tried, it booted up, then allowed me to load a game but then none of the controls worked meaning I couldn’t press Start or any buttons to start the game, forcing a reset.

The card used was a 128GB SanDisk brand microSD card with ArkOS on it. The same issue persisted with both the original microSD card and the new SanDisk microSD card.

All I had done was connect to my own WiFi network and updated ArkOS using the WiFi updater. The process seemed to complete ok at the time. 

What I have noticed, interestingly, is that if I fully charge the device to 100%, then it seemed to work perfectly with either card until I either:
- Restarted the device
- Shut down the device and then booted up again
- Reached around 30-35% battery life

I would continuously require to “hard” reset or shutdown the device using the physical buttons when these issues or freezes occurred. Is there a way to safely restart or reset the device if freezing occurs?

Occasionally I could reach a lower percentage battery, but ONLY if I was playing a game and not using the menu system at the time.

I tried a test where I charged the unit to 100% and put it into sleep mode. The following happened when waking the unit from sleep the following day:
- Battery on unit was showing 55% by morning
- I played MegaDrive games on the unit for almost 1.5 hours.
- Battery went down to 40%
- I tried another game (Dreamcast), but game froze on load up forcing a hard restart using Reset button
- Unit restarted first time BUT randomly the Wifi symbol had disappeared, battery down to 35% and menus began to stutter and pause momentarily
- I put the unit to sleep again
- Woke the unit from sleep and the same freezing and pauses on menus occurred
- Unit felt warm on back of unit, so I shut down the unit completely using menu system
- Switched on unit again, start up fine, still no wifi symbol and battery strangely back to 40%
- Menus showing but system completely frozen forcing me to either shutdown or restart
So this was the same behaviour as previously.

As soon as the unit reaches 30-40% battery remaining (whilst in menu system), or I restart at all, the unit starts acting erratically and more often than not fails on the Emulationstation system loading progress bar.

The unit worked perfectly but not consistently enough to make me feel confident keeping the system, so I was advised to return the system for a replacement, which I did.

I wondered if I had done anything wrong or if I could avoid these issues with my new unit? I know it’s possible it was just a hardware fault, but I always used ArkOS, so thought I would check.

Also, as mentioned above, is there a way to safely restart or reset the device if freezing occurs to avoid possible corruption of any new Micro SD card?

Thanks

christianhaitian, 2021-02-19:
Sounds like your unit may have a faulty battery or circuit.  If you still have the original SD card and OS that came with the unit, try using that.  If the same issue occurs there, you have a hardware issue.  If not, try a different SD card in case the one you currently have may have become faulty.  Good luck.

closed 2021-02-19 by christianhaitian

SegaMad, 2021-02-19:
Hi,

Thank you so much for getting back to me.

I still have both of the SD cards and the issue seems to exist on both unfortunately.
Strangely though, the system was working 100% of the time in all instances apart from when it hit 35% battery. Even then, if I was purely gaming at the time, the battery would last until it hit 10% or even lower? Very odd. 

I thought perhaps the card or ArkOS image/firmware had become corrupted somehow due to the freezing or hard resets?

I no longer have the original unit unfortunately. I got a replacement RG351M and this seems to be working well so far, except that the rumble/vibration sequence doesn’t happen on boot up?

I thought perhaps this was Anbernic eventually answering people’s complaints and removing the vibration at start up? Or do you think I am completely unlucky and now have a *2ND* defective unit?

Thanks,

Simon Scott

christianhaitian, 2021-02-20:
There have been reports of some units that have the rumble motor come loose from the board.  If you know how to solder, you can fix this easily yourself by just reflowing the solder on the 2 connection points of the motor.  Otherwise, yeah, you have a little bit of unluckiness happening.  :(

SegaMad, 2021-02-20:
Hi,

Unfortunately, I’m not very technologically inclined and have never soldered, so that wouldn’t be a possibility.

Would continuing to use the unit in this condition cause any further damage? I might be able to live without a rumble motor. If not, I can return (again!). :(

SegaMad, 2021-02-20:
I’m also having an issue where some ROMs (which previously scraped ok on ArkOS using Screenscraper.fr) are no longer being found? 

Is there a fix or just something I’m doing stupidly?