christianhaitian/arkos #261: \sgb not recognised as a system by the scaping menus

AdeCarter, 2021-02-20:
Placing roms in \sgb, while recognised as a system by the front end, doesnt allow me to scrape them as there isnt a platform assigned,

christianhaitian, 2021-02-20:
Yep.  Missing scraping for super gameboy in emulationstation-fcamod.  I'll make sure to add it for the next update.  Just making a note for myself that the Platform ID is 127.  Thank you for reporting this.

christianhaitian, 2021-02-20:
I updated Emulationstation to address this issue.  You're welcome to restore this backup of it and confirm it solves the issue for you as well. See below:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test super game boy scraping.

Let me know if it's successful or not.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6016603/arkosbackup.tar.gz)

AdeCarter, 2021-02-20:
Can confirm fixed via restoring that backuo

christianhaitian, 2021-02-20:
It shouldn't as it's just an updated executable.  Enjoy!

closed 2021-02-20 by christianhaitian