christianhaitian/arkos #262: System Failure - What went wrong?

SegaMad, 2021-02-21:
Hi,

I’ve been using ArkOS with no issues on a new SanDisk 256GB card for the past few days in my RG351M.

I was using the in built wifi to scrape games today. Finished up and restarted the system (properly and using the menu system).

Now the system won’t start at all.

All I’m getting is a screen saying “System Failure” and “booting fail! check kernel and dtb images” followed by another saying “power off...” then the system shuts off.

What has happened here? Has the unit failed or the card?

I took the card out and tested in a card reader and the “BOOT” and “EASYROMS” folders and contents all still seem to be there?

I’m really not having great luck as this is my 2nd SanDisk card and 2nd RG351M with issues.

Any thoughts?

christianhaitian, 2021-02-21:
Try the original SD card that came with the unit and see if that boots without issue.

SegaMad, 2021-02-21:
Hi,

Thanks for getting back to me so quickly. I tried the no brand card that came with the RG351M and it boots up but takes quite a bit longer, but does boot up into the main system.

I can still access both the “BOOT” and “EASYROMS” folders and have attached a screenshot of the contents of the “BOOT” folder. Can you see anything that could be the issue?

As I said, this is now my 2nd RG351M unit (the 1st had a battery issue and this one has a broken rumble unit) and this is now my 2nd SanDisk branded card to fail or do something odd.

Is there a reason this keeps happening? I can access all of the files on the card, so it seems to be something wrong with the firmware (ArkOS) or something becoming corrupted somehow?

What are my options?

[image]

SegaMad, 2021-02-21:
Are there any of the above files out of place or anything easy I can do without reflashing or replacing a card again?

SegaMad, 2021-02-21:
I’ve noticed that the images in the BMPs folder are 461KB each (logo1, logo2, logo3) but the logo file in the main section of the BOOT partition is only 110KB? Is that right?

I’ve tried looking at the card again and noticed that the preview of the ‘logo’ file has disappeared now too, so it’s just a white file but still is 110KB in size, instead of the other files which are 461KB.

Could this be what the problem is?

christianhaitian, 2021-02-21:
Unfortunately I have no answers for you on this.  I and many others use ArkOS on the rg351p and other devices with no issues like this.  It could be the sd card you obtained is faulty but it sounds like the unit itself is faulty already since the rumble motor doesn't work.  I do have an updated image that I will be releasing shortly.  I doubt that it will help but if you plan to reimage your card, you can use the new image.  Here's link to the new image if you want to try that.  https://drive.google.com/file/d/1UEcQffIgL7B2ctHPsi_GjnefzBHCq56j/view?usp=sharing

christianhaitian, 2021-02-21:
I should add that in looking at the files in the boot folder, it looks like some files got corrupted and as part of the fix, it created some fsck files that it found were corrupted but didn't know what to do with them.

SegaMad, 2021-02-21:
Hi,

Thanks for getting back to me. Yes, I totally understand. From the posts I’m seeing on Reddit etc, it really does seem to be the new RG351M units that are having the issues rather than the P?

Ok, so which files should not be in the folder (as in the extra fsck files)? Do you have a list or screenshot of what should be in the BOOT partition? Sorry, just so I know for the future.

So would you say it’s definitely an SD card issue or perhaps a unit issue? It’s just really frustrating as this is the 2nd ArkOS image I’ve used on a 2nd Sandisk card also. Just trying to work out what keeps going wrong and I really appreciate your feedback.

Is there a way to fix this current card/firmware by fixing that ‘logo’ file and getting rid of the extra fsck files or is that a lost cause?

christianhaitian, 2021-02-21:
You can open the image file with 7zip and look at the boot partition to see what the default image files are.  The ones that are not normally there are the ones that end in .rec.  Sorry for your troubles but not much I can do with hardware issues on my end.

closed 2021-02-21 by christianhaitian

SegaMad, 2021-02-21:
Thanks for getting back to me and really do appreciate your help!

Is it possible that faulty hardware can cause corruption like this then? Just so I know before returning my 2nd unit.

christianhaitian, 2021-02-21:
Anything is possible when hardware is not correct.  The 3rd partition is a partition only Linux can read called root.  it's possible that it hasn't gotten corrupted.

SegaMad, 2021-02-21:
Thanks!

I’ve just unzipped the linked file you sent and it has an .img file as you said within it. How do I view or expand the contents without flashing it to a card? Sorry, I’m not as up on certain technical aspects.

christianhaitian, 2021-02-21:
With 7zip