christianhaitian/arkos #263: Ark os wont format and cant add any files

iCheetle, 2021-02-22:
When i add files they wont stay but when i turn the system on they are not there it just the same thing for example i go and rename a file or delete it and i restart the system and it goes back to where it was before and i cannot format my SD card either.

christianhaitian, 2021-02-22:
Sounds like an issue with the SD card.  It’s as if it is preventing writing.

iCheetle, 2021-02-22:
could you help

christianhaitian, 2021-02-22:
I can't fix a hardware issue like a possibly bad SD card.  What SD card is it and where did you get it from?

iCheetle, 2021-02-23:
the one that comes with the rg351p

christianhaitian, 2021-02-23:
Those are known to fail quickly and cause issues.  That's why it's not recommended to use the default card that comes with the rg351p unit.  Get an authentic Sandisk or Samsung SD card and install the latest image on that.

closed 2021-02-23 by christianhaitian