christianhaitian/arkos #264: Some ports and some systems crash, others do not

multicam19, 2021-02-23:
### Describe the Issue (If applicable)
I am unable to run PSX, N64, Dreamcast games, and ports.  SNES, NES, GB, GBC, and GBA games still work.  I first noticed the issue after beating Super Mario 64, getting stuck on the "The End" screen, and resetting the RG351M.  I tried to go back into SM64 and it acted strangely: I hear the game start up and register my inputs (I hear all the normal sound effects that you would expect when starting the game.  I can even load my game and start playing, but I see nothing).  The only thing on screen is a yellow blinking underscore in the top left corner.  It is still behaving exactly like this.

I figured something was wrong with my Super Mario 64 port and exited using Select and Start.

Other ports, including Doom, Quake III, Half-Life, and others, don't behave the same way.  They fade to black as if they're about to load, then crash to the menu where I selected them from.  

What's weird is that N64, PSX, and DC games do the same thing, but SNES, NES, GB, GBC, and GBA work fine as always.

I figure it has something to do with Super Mario 64 and/or Retroarch vs Retroarch32, but I don't know what could have changed to make it start acting this way.

### How can the issue be reproduced? (If applicable)
^any time I try to play a port or game from the systems listed above.

### What device are you using?
RG351M with a 256GB Samsung EVO SD card.

### What version of ArkOS are you on?
1.5 with the 02 20 2021 update.

### Anything else you'd like to include that may help to review this issue or feature request?
One port I've tried does work- Cave Story.  The others I've tried do not work.

Thanks in advance for any help!

christianhaitian, 2021-02-23:
When you say super mario 64, do you mean the port or were you playing it through the n64 emulator?

christianhaitian, 2021-02-23:
If so, maybe it may have caused an issue with the system.  As I don’t support the port, I don’t know what issues it may cause.

multicam19, 2021-02-23:
I’m referring to the actual port, not the rom. 

There appears to be something wrong with Retroarch32. When I open Retroarch by itself I get the normal menus but Retroarch32 crashes back to the menu like PSX, N64, and DC games and most ports. 

Is there a way to do some kind of reset or restore that doesn’t involve starting from scratch? I really don’t want to lose all my saves and settings!

christianhaitian, 2021-02-23:
You're welcome to restore this backup of retroarch32 and confirm if it solves the issue for you or not. See below for instructions:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test retroarch32
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6030923/arkosbackup.tar.gz)
.

Let me know if it's successful or not.

multicam19, 2021-02-24:
Nope, didn't work :/ all of the same symptoms exist.

I did notice something weird... in the ports list, if I try 3-Demon, it starts up DOOM 2 which runs flawlessly.  I never tried 3-Demon before I had this issue because I've never heard of it, so I don't know if this would have happened before.  Not even sure if this is related.

Thanks for your continued help.  I really appreciate it

christianhaitian, 2021-02-24:
That's not related.  You must have scraped the ports folder and it's named the various ports incorrectly.

christianhaitian, 2021-02-24:
Unfortunately, I have no other suggestions on how best to resolve your issue with retroarch32.  You'll probably need to backup your easyroms partition, flash the latest version of ArkOS and copy your roms back.

multicam19, 2021-02-24:
Okay thanks for your help.

As a final note, I did notice this in the gamelist.xml file:

<game>
		<path>./Doom 2.sh</path>
		<name>3-Demon</name>

I have no idea why it would have changed that but it explains why 3-Demon opens Doom 2.  Anyway, I'm sure you're right, at this point I'll just call it and start from scratch.

multicam19, 2021-02-24:
game
		path . /Doom 2.sh /path
		 name 3-Demon /name

(deleted the carrots so it would show up in this comment)

christianhaitian, 2021-02-24:
Ok.  Good luck

closed 2021-02-24 by christianhaitian