christianhaitian/arkos #265: Remapping controls for STAND-ALONE-RICE

eckert2021, 2021-02-24:
Hello!

it is undeniable that the performance you get out of the stand-alone nintendo 64 emulator (RICE) is miles ahead of what you can get from the Retroarch cores.

I personaly don’t have a problem playing in a emulator which you cannot set or configure anything as long as it plays well. Even the control scheme mapped for that emulator is fine, except for the fact button Z being on the R3 position. That really messes up with basically every game you make a good amount of use of it. 

I was wondering if somehow I could remap the control layout only to switch the button Z (R3) with button C-UP (L2). Just that...

christianhaitian, 2021-02-24:
You can do this manually.  Assuming you're device is a RG351P/M, See [here](https://github.com/christianhaitian/arkos/wiki/Key-codes-and-Global-Hotkey-config-file-locations----RG351P#global-hotkey-config-file-locations) for more info on this.

closed 2021-02-24 by christianhaitian

eckert2021, 2021-02-24:
Thank you so much!