christianhaitian/arkos #266: arkOS zip file

djmmimer, 2021-02-25:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/RockRiotMenu) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

### What device are you using?

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request?

djmmimer, 2021-02-25:
I've download the zip file from either MEGA or GDrive sites, tried to unzip it and it seems like it stop at 29% and gives me these messages when the file is being process:
1 Can not set length for output file: There is not enough space on the disk : E:\Mine\ArkOS_RG351P_v1.5_02202021(good)\ArkOS_RG351P_v1.5_02202021.img
2 E:\Mine\ArkOS_RG351P_v1.5_02202021(good).7z
3 ArkOS_RG351P_v1.5_02202021.img
4 There is not enough space on the disk.

I have plenty of space over 100 GB. so why am I getting this and it hang/freeze at 29% for both files at both sites. Is there another site I can get a clean download of the file?

christianhaitian, 2021-02-25:
Where are you expanding the image onto?

djmmimer, 2021-02-25:
I'm explaining it to my harddrive where it was downloaded to.

christianhaitian, 2021-02-25:
is your harddrive formatted to fat32?  It seems the error is stating it can't expand the image as the file system can't support the size of the expanded image.

djmmimer, 2021-02-25:
Oh I'll have to redirect it to another hard drive that's not fat32 correct?

christianhaitian, 2021-02-25:
Yes.  fat32 has a max single file size of 4GB.  The image is 8GB in size when expanded.  If this will be an issue, you can try 351elec which is much smaller.  Good luck.

closed 2021-02-25 by christianhaitian