christianhaitian/arkos #272: Amiberry's Button Mapping is not working as it should

Humungus303, 2021-02-27:
Hello there!

I recognized that the Buttonmapping / Mouse to Analog - Function  of Amiberry is not working as it should.
Only the Mainbuttons and the D-PAD are working with mapping and the North/South/East/West is not correct dressed
(North is Right)

And the Analog Sticks cannot be used as a mouse as normal.
though the Select/Start Button can be used for quickmeu they cannot be mapped also.
As the Shoulder Buttons nor the AnalogStick Buttons.

Thank's for your attention! And many more thank's for building this all.

christianhaitian, 2021-02-27:
See the Amiga section of the global hotkeys in the FAQ [here](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS) for more information on how to adjust these settings yourself.  It will be up to you to adjust the keys for that emulator for the game(s) you want to play.

closed 2021-02-27 by christianhaitian

Humungus303, 2021-02-27:
Hello Christian,
thank you for your quick reply. I am quiet familiar with all the emulating settings that has to be done to make things right andget them working correctly. But this one gives me a headache. Amiberry as it is  a standalone operation should receive the right OpenSimHardware OSH PB Controller.cfg. But it doesn't.

For example. the current version of amiberry under emulec is working correct with all the buttons acceptable for mapping. Mouse also works as it should. Can map everything on both sticks as i want, But not under ArkOS.

WHDLoad and adf. lha everything works great on ArkOS. Everything is superb. Only this tiny issue is making me sick :)
If you are the developer of ArkOS. Thank you so much for this great system. You have made many people happy !
Cheers

christianhaitian, 2021-02-27:
Unfortunately, I have no better solution for this for Amiberry.  I have heard that the retroarch emulator seems to work better than amiberry for many of the games.  Perhaps try that one.

Humungus303, 2021-02-27:
Yes the PUAE is doing quiet a good job but amiberry has the much better scaling and more Options for WHDLoad and other stuff if you want the better Amiga expirience..anyway. thanks for your time.much appreciated! i will try something else...don't want to switch system. your ArkOS is wonderful with the 351m

input_driver = "udev"
input_device = "OpenSimHardware OSH PB Controller"
input_vendor_id = "4617"
input_product_id = "12544"
input_b_btn = "0"
input_y_btn = "3"
input_select_btn = "22"
input_start_btn = "21"
input_up_btn = "h0up"
input_down_btn = "h0down"
input_left_btn = "h0left"
input_right_btn = "h0right"
input_a_btn = "1"
input_x_btn = "2"
input_l_btn = "19"
input_r_btn = "20"
input_l2_btn = "25"
input_r2_btn = "26"
input_l3_btn = "23"
input_r3_btn = "24"
input_menu_toggle_btn = "2"
input_enable_hotkey_btn = "7"
input_exit_emulator_btn = "6"

Humungus303, 2021-02-27:
oops sorry for the CopyPaste .cfg
:) have a nice weekend!

Humungus303, 2021-02-27:
Both emulators works fine for all Games by the way. It's just that most users don't update their WHDLoad.cfg's or turning on different shaders in PUAE resulting in stuttered sound and framedrops. And always use the right version of the Kickstart roms for each Game.

PS: It seems that both controller.cfg are the same. i have no clue what to do next. maybe midwan from amiberry will know.
Greetings!