christianhaitian/arkos #277: Ir-glupen64 just doesnt’t work

eckert2021, 2021-03-02:
RG351P
ArkOs 1.5 (02272021)

I have been trying to test out which emulator performs best overall with the Nintendo 64.

Considering the latest update of ArkOs, I’ve come across with the following available emulators:

Standalone-rice
Standalone-GLideN64
Ir-paralellel (32/64 bit)
Ir-mupen64plus_next
Ir-glupen64

However, out of these 5 available options, it seems that when choosing the glupen64 core I cannot make any game to work - they won’t even start/boot initially.

It’s worth mentioning that I did try to update and re-download this core directly from the retroarch library, but it still didn’t work.

If anyone else is having this issue or can tell me if this is a completely normal fenomena, I appreciate.

christianhaitian, 2021-03-02:
That emulator doesn't work well.  Even though it's available, it hasn't been updated since 2016.  I wouldn't bother with it.  For the best emulation performance, the standalone emulators are best.  If you need to change the button mappings, you'll have to do so manually using the information [here](https://github.com/christianhaitian/arkos/wiki/Key-codes-and-Global-Hotkey-config-file-locations----RG351P)

closed 2021-03-02 by christianhaitian

eckert2021, 2021-03-03:
Ok! Thanks for the fast reply