christianhaitian/arkos #278: Game selection menu issues

vimstar, 2021-03-03:
I just updated and now I can no longer see any selection indicator when browsing games to play. It is impossible to tell which game is selected. 

Also, is there any way to revert to the old game menu setup as I liked that layout better than the new RetroPi style system selection screen.

christianhaitian, 2021-03-03:
I'm not understanding the issue.  Can you post a picture of the issue?

vimstar, 2021-03-03:
In the attached picture you will notice the blue game list has not indicator for which game you have selected. There are no indicators for what is selected on any of the screens that have a dark blue list.

By the way, awesome job on ArkOS! 

[image]

christianhaitian, 2021-03-03:
Looks like your missing the theme.  Do you have a theme folder in your roms directory?  If so, does it have the themes listed [here](https://github.com/christianhaitian/arkos/wiki/Themes)

vimstar, 2021-03-03:
Yes, I do have the themes folder in the roms directory and it contains all the files listed.

[image]

christianhaitian, 2021-03-03:
Try changing themes to see if that fixes the issue.

christianhaitian, 2021-03-03:
You can do so from the start menu

vimstar, 2021-03-04:
The other themes appear to be working. It seems its only a problem with the es-theme-nes-box theme. When I select that theme I get what you saw in the screen shot. Ill try downloading that theme and replace the pre-installed one to see if that fixes the issue. Thanks!

christianhaitian, 2021-03-04:
Make sure to get it from [here](https://github.com/Jetup13/es-theme-nes-box) so it's formatted properly for ArkOS.

closed 2021-03-09 by christianhaitian