christianhaitian/arkos #279: NEW ADAPTED THEME FOR ARKOS ( based on EPICNOIR )

CodyV59, 2021-03-04:
Hi, Christian i work on a new theme based on epic noir especially adapted for ArkOS, if you want to test and add it you're welcome

her the link to download it   :  https://drive.google.com/file/d/1h9sZ73Kelknp15vMp6ROcZ-j-EiUzMeq/view?usp=sharing

Thanks 
Cody

CodyV59, 2021-03-04:
added PICO-8 , SNES-MSU1 and others missing

christianhaitian, 2021-03-04:
Thanks, however, there's a version of epicnoir adapted for ArkOS [here](https://github.com/Jetup13/es-theme-epicnoir).  Were you aware of this?  The wiki has a list of themes that have been adapted for ArkOS by Jetup13.  Below is the direct link to it:

https://github.com/christianhaitian/arkos/wiki/Themes

CodyV59, 2021-03-04:
Epic noir on arkos is not complete
Not all the system and i've Made somerhing différent with bigger image and
video on menu etc....

christianhaitian, 2021-03-04:
The one from the wiki is more up to date with systems than what's included with ArkOS.  I only update the default theme.  It's expected that the other themes are updated at user preference.  There's also a theme's updater that can do it as well from that theme link.  Either way, thanks for sharing.  If you have any previews of it and a github where you house the work, I can post a link to it from my wiki as an alternate version.

CodyV59, 2021-03-06:
added all icons for arkos option menu here

https://drive.google.com/file/d/1vWEKJs1ZjNkc4t02U766BrFitaN2Vkaf/view?usp=sharing

CodyV59, 2021-03-06:
[image]

CodyV59, 2021-03-06:
[image]
[image]
[image]
[image]

christianhaitian, 2021-03-06:
Wow that's impressive.  How does it look in fullscreen mode?

christianhaitian, 2021-03-06:
Also your google drive link doesn't seem to be public.  It's stating that I have to request access.

CodyV59, 2021-03-06:
and now it's OK for the links ???

in full screen it's OK too i've adapted for both

christianhaitian, 2021-03-06:
The google links work fine now.  Can you create git repository for your work?  It would be cool to be able to add this to the themes downloader if TadMSTR agrees to add it there.

CodyV59, 2021-03-06:
[options.zip](https://github.com/christianhaitian/arkos/files/6095481/options.zip)

CodyV59, 2021-03-06:
the option icon must be in arkos on ( roms/options/images/ )  

 ( home/ark/.emulationstation/recovery/retropie )

TadMSTR, 2021-03-06:
just open an issue on ArkThemes github when it's ready and I can add it to ArkThemes

CodyV59, 2021-03-06:
have you the link page for the ArkThemes github ?? i can't find it ???

TadMSTR, 2021-03-06:
https://github.com/TadMSTR/ArkThemes

CodyV59, 2021-03-06:
Ok done ;)

CodyV59, 2021-03-06:
possible to take UI screenshot on ArkOS ??

christianhaitian, 2021-03-06:
Not natively.

krishenriksen, 2021-03-06:
This one should be able to record your screen.
https://github.com/Cebion/packages/releases/download/1.0/record.zip

CodyV59, 2021-03-06:
not work for me ???
i've put the package on tools and i see the capture in the option .... i launch it but it return to the option screen 

how does it work ??

thanks

CodyV59, 2021-03-06:
added a icon option for arkthemes et recordscreen
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

CodyV59, 2021-03-12:
UPDATE OF THE EPIC CODY THEME.

Add EASY RPG, SNES HACKS, SUPER GAMEBOY systems

https://drive.google.com/file/d/17aTO5CZjgtqPFcgwxwEbW8zKv7EWfWD-/view?usp=sharing
or
https://1fichier.com/?04d4qlu1umron7r1s1go

duskgao, 2021-03-12:
> UPDATE OF THE EPIC CODY THEME.
> 
> Add EASY RPG, SNES HACKS, SUPER GAMEBOY systems
> 
> https://drive.google.com/file/d/17aTO5CZjgtqPFcgwxwEbW8zKv7EWfWD-/view?usp=sharing
> or
> https://1fichier.com/?04d4qlu1umron7r1s1go

Hi~ I found famicom is same with fds in your theme...Could you fix it?

and I also add many system to arkos in issue #281 ,many of them lack theme pics...How about add more pics for these system?

thx for your great work!

CodyV59, 2021-03-12:
Yes i will fix iy for fds and famicom
I'm Open to add others system too 
if Christian add it 
Make me the list who miss

duskgao, 2021-03-13:
OK,just compare your theme and arkos system

1.Exists in arkos but not in your theme
c128
pcfx
odyssey2
ti99
uzebox
vmu
sufami

2.diff dir in arkos and your theme
dosbox,   your theme is “pc”  , in arkos is "dos"

3.not in arkos but exists in otheros
channelf
moto
pc88
satellaview
videopac
vic20
gb2player
gbc2player
sc3000
megadrive-japan
genh
gbah
gbch
gbh
nesh

CodyV59, 2021-03-16:
UPDATE EPIC CODY THEME

ADDED SYSYEMS
c128
pcfx
odyssey2
ti99
uzebox
vmu
sufami

and DOS modified 

https://drive.google.com/file/d/17aTO5CZjgtqPFcgwxwEbW8zKv7EWfWD-/view?usp=sharing

CodyV59, 2021-03-21:
Hi Guys .... DOS ans favorites fixed

THEME EPIC CODY V2.1 FULLSCREEN

https://drive.google.com/file/d/12AeEBGjs_chOkT-ivuLWvvURQY-9eb71/view?usp=sharing

THEME EPIC CODY V2.1 ARKOS HEADER

https://drive.google.com/file/d/1MloFAaBOsDVKljv_fY_jy4KXB3O_5EqV/view?usp=sharing

christianhaitian, 2021-03-21:
Thank you for confirming that you will create a proper git repository for your work.  I will gladly link to it upon completion.  Thanks again.

closed 2021-03-21 by christianhaitian