christianhaitian/arkos #281: More systems may emulate by arkos?

duskgao, 2021-03-09:
I am so happy that arkos has a separate supergb system. Now,I checked out the systems that not  in arkos, but have been emulated in other OS, for your reference. I hope they would appear in arkos soon...I had login into arkos from sftp,but I can't find es_system.cfg & getcores.sh,so I can't help to add these content myself :(

here is the list:

1.amigacdtv , emulated by puae_libretro,exists in arkos

2.channel F, emulated by freechaf,exists in arkos

3.FBneo,  there is no separate fbneo contents in arkos like other os

4.gx4000, should emulated by cap32_libretro? also in recalbox 7.1.1

5.MO5/TO7, emulated by theodore_libretro,exists in arkos

6.pc88, emulated by quasi88_libretro,exists in arkos

7.Palm OS, emulated by mu_libretro,exists in arkos

8.satellaview , use snes9x_libretro

9.videopac ,use o2em_libretro

10.vic20, use vice_xvic_libretro

11.gb2player/gbc2player just a category for tgbduel 

12.apple2/apple2gs in recalbox 7.1.1,but a standalone linapple also work in Retro-Arena (all these standalone emulators can be found at  https://github.com/Retro-Arena/ARES-Binaries/tree/master/buster/odroid-n2/emulators)

13.oric in recalbox 7.1.1 (standalone emulator oricutron in Retro-Arena)

14.Coco/Dragon32/Dragon64: standalone emulator xroar work in Retro-Arena

15. Macintosh: standalone emulator minivmac work in Retro-Arena

16.java, not in arkos but there is java core(lr-freej2me) in Retro-Arena could you test this? (https://github.com/Retro-Arena/ARES-Binaries/tree/master/buster/odroid-n2/libretrocores)

17.mess core not in arkos but  Retro-Arena has mess/mess2016 cores ,so these systems are in Retro-Arena.(https://github.com/Retro-Arena/ARES-Binaries/tree/master/buster/odroid-n2/libretrocores)
Supervision
Megaduck
Tiger Game.com
Adventure Vision
Arcadia-2001
Super A'can?

How can I help to add these system to the future arkOS update?

christianhaitian, 2021-03-09:
This is quite a large request.  Not sure how much of these I will be able to add as for instance, the Apple emulator requires a keyboard to function last I recalled.  Macintosh, would it not need a keyboard as well?  I focus on emulators that can function without the need for a keyboard.  FBNEO is already included.  It's under the Arcade section as identified in the ArkOS Emulation and Ports section of the wiki [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#arcade).  The es_systems.cfg is located in the /etc/emulationstation folder.  

All Libretro cores are installed in /home/ark/.config/retroarch/cores (64 bit) and /home/ark/.config/retroarch32/cores (32 bit).  I typically have all standalone emulators installed in their own separate subfolders within the /opt folder.  You're welcome to test and if successful, report what you did so I can test and review for possible inclusion.  As a reminder, I will only include emulators that do not need to have a keyboard to use.

Thanks,

duskgao, 2021-03-09:
thx @christianhaitian for your help. 

I used edit ee3.6 's es_system.cfg & getcores.sh to add new system, and it works.I don't know if arkos is similar. I will try to modify the arkos’s es_system.cfg, if it works, I will report here. :D

christianhaitian, 2021-03-13:
I've reviewed and many of the entries seem ok.  I may just need to fix some formatting at most.  Last I recalled, mesen did not function well on this platform.  I will probably not include that.  For the new emulators such as ChannelF, moto, vic20, etc. I will need to update the Emulators and Ports section of the wiki for them so people know how to properly get them running.  Do you have suggested entries you can submit a PR for?

duskgao, 2021-03-13:
Yep ,I can help to submit a wiki PR for you! Just tell me which system you would include in arkos :)

christianhaitian, 2021-03-13:
amigacdtv
channelf
gb2player
gbah
gbc2player
gbch
gbh
genh
moto
nesh
pc88
satellaview
sc3000
vic20
videopac

duskgao, 2021-03-13:
please check the wiki file if you have time

[arkos_wiki.txt](https://github.com/christianhaitian/arkos/files/6135168/arkos_wiki.txt)

fixed some bug in es_system.cfg
1.wrong command in gb2players & gbc2players
2.add m3u to pc88
3.made blusmax for sc-3000
[es_systems.txt](https://github.com/christianhaitian/arkos/files/6135198/es_systems.txt)

SuperBadger77, 2021-05-04:
Still angling for a BBC Micro emulator :-)

duskgao, 2021-05-05:
> Still angling for a BBC Micro emulator :-)

maybe advmess had a chance...

SuperBadger77, 2021-05-05:
> > Still angling for a BBC Micro emulator :-)
> 
> maybe advmess had a chance...

If you get advmess working \ installed - I will give it a go. On the RetroPi Blue Motion Unified image it was working - so lets hope so :-)

SuperBadger77, 2021-05-15:
Is Advance Mess up and running?

duskgao, 2021-05-15:
> Is Advance Mess up and running?

Unfortunately,bbc not working...
but I test many systems with lr-mess..some of them working,some of them can‘t control

SuperBadger77, 2021-05-15:
> > Is Advance Mess up and running?
> 
> 
> 
> 
> 
> Unfortunately,bbc not working...
> 
> but I test many systems with lr-mess..some of them working,some of them can‘t control
> 
> 

Anything I can do to help? What sort of error did you get? Is lr-mess already on the latest version? If so can I just ssh onto the machine and try executing lr-mess?

duskgao, 2021-05-16:
> Anything I can do to help? What sort of error did you get? Is lr-mess already on the latest version? If so can I just ssh onto the machine and try executing lr-mess?

No, I don’t think we can do much at the mess core. You can find the core here (https://github.com/Retro-Arena/ARES-Binaries/tree/master/bionic/jetson-nano/libretrocores)

Due to complex settings ,all the _cass,_flop,_cart not work for me. This is the situation in BBC Micro. Especially for pockstat(Sony PocketStation), BIOS does working with controller,but no where to mount the .gme file.

this is the system list I had tested. They all showed at least a BIOS screen on arkos,some of them are fully palyable.

supracan (slow)
arcadia (Partial control)
apfm1000 (Unable to control)
gx4000 (slow)
megaduck (fully palyable)
gamecom  (Unable to control)
scv  (Unable to control)
advision (fully palyable)
astrocde  (Partial control)
gmaster (fully palyable)
gamate (fully palyable)
gamepock (Partial control)
gameking (fully palyable)
gamekin3 (fully palyable)
gp32 (slow)
pv1000 (fully palyable)
vc4000  (Partial control)
crvision  (Unable to control)
microvsn  (Unable to control)
cdimono1  (Unable to control)
sv8000  (Unable to control)
fmtmarty (slow)
apple2gs (Unable to control)
myvision  (Unable to control)
tutor (Unable to control)
studio2  (Unable to control)

I personally think that emulating these retro keyboard devices on a handheld is not easy because of the ra‘s retropad remapper setting.

Chiffon-Pudding, 2021-05-17:
Hello.
I have successfully built and run "MKXP"(https://github.com/Ancurio/mkxp) on ArkOS on RG351P.
https://www.reddit.com/r/RG350/comments/kuy4d2/mkxp_with_arkos_on_rg351p/
This allows us to run RPG Maker XP/VX/VXAce games.
It appears to work well, although there are some somewhat heavy running, some graphical bugs and font problems (by default, problems with Japanese, etc.), and a few games that don't work (e.g. when using the Win32API).
However, each game requires a mkxp.conf file to describe its settings.
I have yet to figure out how to make this work...

I was able to build "MKXP-Z"(https://github.com/night-burst/mkxp-z), but when I try to run it, I get an "Error initializing SDL: No available video device" error and it immediately quits.
"NW.js" (for RPGMaker MV, Ver. 0.23.7... the only version with Arm64 build https://github.com/LeonardLaszlo/nw.js-armv7-binaries/releases/tag/v0.23.7) does not output anything and quits immediately too.
[image]

duskgao, 2021-05-18:
> Hello.
> I have successfully built and run "MKXP"(https://github.com/Ancurio/mkxp) on ArkOS on RG351P.

Great!Can't wait to test it! 

But I think you need to open a new issues to @christianhaitian .So he would find you and deal with your awesome work. :)

Chiffon-Pudding, 2021-05-19:
> But I think you need to open a new issues to @christianhaitian .So he would find you and deal with your awesome work. :)
Okay. I'll do it that way.
The work may take several days. 
Thanks!

Chiffon-Pudding, 2021-05-29:
Okey, but @christianhaitian didn't want us to create an issue about the new port, so I created an issue at https://github.com/krishenriksen/AnberPorts that he linked to: krishenriksen/AnberPorts#51

tylibra2, 2021-11-23:
@duskgao - It managed to emulate the Palm OS, I put the 3 files it asks for but nothing appeared. Only the buttons at the bottom appeared, but it didn't play any games. But, I haven't tested it with all the bios yet. I did it for RetroArch. Did you emulate? ChaiLove got just Flappy Bird emulator and a snake game.

SuperBadger77, 2021-11-23:
> No, I don’t think we can do much at the mess core. You can find the core here (https://github.com/Retro-Arena/ARES-Binaries/tree/master/bionic/jetson-nano/libretrocores)
>
> Due to complex settings [...]

@duskgao is the lr-mess or lr-mess2016 included on the current build of ArkOS? If so I can take a look at seeing if I can get it to work, or get some help in. Could you talk me through the steps to get to where you got to with it and I can investigate further?

I've been trying to get the BBC Micro added to RetroArch \ Lib Retro. See the following posts:
https://github.com/libretro/RetroArch/issues/13108#issuecomment-962507678
And Jeremy raised this issue upstream as well: https://github.com/stardot/b-em/issues/162
These guys may also be able to help.

SuperBadger77, 2021-11-23:
> No, I don’t think we can do much at the mess core. You can find the core here (https://github.com/Retro-Arena/ARES-Binaries/tree/master/bionic/jetson-nano/libretrocores)
>
>
>
> Due to complex settings [...]

Also found this thread on RetroPie: https://retropie.org.uk/forum/topic/29402/bbc-micro-acorn-electron-emu/178?lang=en-GB

christianhaitian, 2021-11-23:
lr-mess is available.  If you don't have it already downloaded, just download it through retroarch

christianhaitian, 2022-11-19:
Some of these emulators have been incorporated and the rest may or may not over time.  Closing this issue as interest in adding the rest may come overtime.

closed 2022-11-19 by christianhaitian