christianhaitian/arkos #282: Wifi card not detected after 3/8/21 update

morden666, 2021-03-09:
I updated to arkOS 1.6 last night and now my wifi card is not detected.
Please find a way to fix this.

christianhaitian, 2021-03-09:
Yes.  The new kernel update affected wifi cards that use the RTL8812 driver.  What is your console and wifi card?

christianhaitian, 2021-03-09:
Ok.  This should hopefully solve it.

1.  Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3.  Once completed. Reboot the device, the device will reboot again and you can test to see if your wifi works again.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6108815/arkosbackup.tar.gz)

christianhaitian, 2021-03-09:
Did you unplug and replug your wifi?  Also, did you try to remove your ssid and re add it?

morden666, 2021-03-09:
I did try rebooting, yes. The wifi adaptor model # Archer T2U Nano

christianhaitian, 2021-03-09:
That should have worked.  It fixed it for someone else with that specific adapter.  Can you try the steps again above and reboot twice afterwards?

christianhaitian, 2021-03-09:
Actually, use this file instead.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6109276/arkosbackup.tar.gz)

christianhaitian, 2021-03-09:
Did you try the second attachment I posted?

christianhaitian, 2021-03-09:
Is the unit pictured in the link below same as yours?
https://www.tp-link.com/en/support/download/archer-t2u-nano/

morden666, 2021-03-09:
Yes. 
[image]

christianhaitian, 2021-03-09:
With the second attachment, when you rebooted the unit, did it reboot 1 more time on it's own as it was entering Emulationstation?

morden666, 2021-03-09:
No I had to reboot it on my own the second time. I actually used the shutdown option.

christianhaitian, 2021-03-09:
Hmmm....Try to restore this one and let me know what happens.
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6109804/arkosbackup.tar.gz)

morden666, 2021-03-09:
Restored using said fail. Rebooted. Did not reboot a second time going into Emulationstation. Still no wifi. Tried another reboot. No dice. Tried doing shutdown then reboot, no dice. Maybe I'll just buy another adaptor lol.

christianhaitian, 2021-03-09:
Sorry.  I don't know what the issue is then as I rebuilt the driver from the previous kernel which should have worked as it did for someone else.  Good luck.

closed 2021-03-09 by christianhaitian

christianhaitian, 2021-03-09:
Odd but ok.  Happy it finally works!