christianhaitian/arkos #285: Feature request: MAME version selection

zomgugoff, 2021-03-12:
In the Emulator Settings, there is no option to change the default emulator for MAME. Currently, there appears to only be support for directories that have MAME2003 and MAME2010, despite other versions being available for download. Can we get either a setting to choose which MAME to run or support for more MAME directories?

duskgao, 2021-03-12:
maybe this scriptcan help?  @christianhaitian 

	<system>
		<name>mame</name>
		<fullname>Arcade - MAME</fullname>
		<path>/roms/mame/</path>
		<extension>.zip .ZIP .7z .7Z .chd .CHD</extension>
		<command>sudo perfmax; /usr/local/bin/retroarch -L /home/ark/.config/%EMULATOR%/cores/%CORE%_libretro.so %ROM%; sudo perfnorm</command>
		   <emulators>
		      <emulator name="retroarch">
		 	<cores>
		 	  <core>mame</core>
		 	  <core>mame2000</core>
		 	  <core>mame2003</core>
		 	  <core>mame2003_plus</core>
		 	  <core>mame2010</core>
		 	  <core>mame2015</core>
		 	</cores>
		      </emulator>
		      <emulator name="retroarch32">
		 	<cores>
		 	  <core>mame2003_xtreme</core>
		 	</cores>
		      </emulator>
		   </emulators>
		<platform>arcade</platform>
		<theme>mame</theme>
	</system>

christianhaitian, 2021-03-12:
What’s the benefit of the other Mame emulators compared to what’s already available and the fbneo and fbalpha emulators?  I haven’t found many arcade games that play better on Mame vs. fbneo or fbalpha except for Hook.

zomgugoff, 2021-03-13:
If MAME isn't useful, I have to question why ES was configured to recognize 2 different MAME folders. Really, just adding the core selection like duskgao proposed is fine.

christianhaitian, 2021-03-13:
This doesn’t answer the question. What is the benefit of adding mame2000, mame2003_plus, and mame2015?  Have they been tested through Retroarch to see how they perform?

christianhaitian, 2021-03-18:
Closing as there's been no further feedback on the benefit that the additional mame versions bring.

closed 2021-03-18 by christianhaitian