christianhaitian/arkos #286: Update error

fabiocattaneo, 2021-03-12:
I've tried performing the latest OTA update. At the end of the process a message warned me that an error occured, to try again.
Trying again it says no more updates are available.
I've opened the log file and this is what I've found:

"Copy correct updated ES for easyrpg scraping fix
'/usr/bin/emulationstation/emulationstation.header' -> '/usr/bin/emulationstation/emulationstation'
--2021-03-12 15:49:01--  http://gitcdn.link/repo/christianhaitian/arkos/main/03082021/newkernelnmodndtb02032021.zip
Resolving gitcdn.link (gitcdn.link)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘gitcdn.link’

The update of your kernel couldn't complete because the kernel package did not download correctly.
Please retry the update again."

How can I fix it?

christianhaitian, 2021-03-12:
Thanks for bringing this to my attention.  Please try again in about an hour from the time of this response and let me know if the update completes successfully.  It should be for the new kernel.

fabiocattaneo, 2021-03-12:
"No more updates available. Check back later"

christianhaitian, 2021-03-12:
Sorry, I forgot to push the changes to the github.  🤦‍♂️  Please try again.

christianhaitian, 2021-03-13:
Have you had a chance to try this again?

fabiocattaneo, 2021-03-13:
I've just retried, a couple of minutes ago and I'm happy to report the Kernel and the modules have been successfully updated :-)
Thanks for the fix!

christianhaitian, 2021-03-13:
Yay!  Thanks again for reporting the issue and confirming that the issue has been resolved.

closed 2021-03-13 by christianhaitian