christianhaitian/arkos #288: Wifi dongle no working after update to 1.6

tomatoskittles, 2021-03-13:
Hey, I used my wifi dongle (TPLink) to update to ArkOS 1.6. (It has worked up until now).

After the update completed, my wifi dongle no longer works.

The light on the dongle itself no longer turns on, when plugged in, or when the system is rebooted with the dongle already plugged in.

Thanks!

christianhaitian, 2021-03-13:
The information in this link should fix your issue.  https://github.com/christianhaitian/arkos/issues/282#issuecomment-793905701

tomatoskittles, 2021-03-13:
I had just decided to search closed issues, and found it :D I'll let you know when it works.

Thanks for the quick response!

tomatoskittles, 2021-03-13:
I followed the instructions, and it worked.

One caveat: Because I was messing around with the settings to try to make it work, before I posted here: additionally I needed to "Enable Wifi" after following those instructions. I must have disabled it before. After that, the dongle works, as expected.

Thank you, again!

tomatoskittles, 2021-03-13:
Following the provided instructions works,

Make sure to re-enable wifi, if you had disabled it.

closed 2021-03-13 by tomatoskittles