christianhaitian/arkos #289: Systemd path units are unable to watch subdirectories of exFAT partitions

ridgekuhn, 2021-03-13:
Creating this issue for referencing in [arklone](https://github.com/ridgekuhn/arklone) repo's codebase. Discovered in #126. Original text follows...

Here's a small path unit demo I've been testing with: [testexfat.zip](https://github.com/christianhaitian/arkos/files/5829472/testexfat.zip). It's set to watch `/roms/nes` and will create a file at `/home/ark/deleteme` when triggered. You should find that if you change the `Path=` setting to `/roms`, it will create the `deleteme` file every time, but if you change it to a subdirectory of `/roms` or any other exFAT partition you mount, it will (maybe) work once, and then never again. I've primarily been working off of fresh installs of the 12-16-2020 image.

Whatever the issue is, I'm beginning to believe it may be specific to exFAT on ArkOS. Anywhere on the `/root` partition appears to work as expected. I also tried reformating the EASYROMS partition to FAT32, and that also appears to work reliably.

_Originally posted by @ridgekuhn in https://github.com/christianhaitian/arkos/pull/126#issuecomment-762180686_

christianhaitian, 2022-02-25:
In case you’re still working on this, I updated the exfat driver for Arkos for all devices except the p/m.  It may have resolved this issue.

ridgekuhn, 2022-02-26:
@christianhaitian Oh, interesting! I haven't been working on it, but would consider updating it since I know at least a handful of ppl use it. I still only own a RG351P; is the new driver built into the kernel, or is it a module I could try to update myself on my RG351P?

christianhaitian, 2022-02-26:
While it is possible to build it with the kernel as I updated the source I use to inckude it, for the rg351p/m, you should be able to update it by grabbing the /home/ark/exfat.ko.351 module file from this package (https://github.com/christianhaitian/arkos/raw/main/02242022/arkosupdate02242022.zip)
And install it using the following process (assuming you’ll be installing the driver from the same location as I did during the update):

```
sudo apt remove -y exfat-fuse
sudo install -m644 -b -D -v /home/ark/exfat.ko.351 /lib/modules/$(uname -r)/kernel/fs/exfat/exfat.ko
sudo depmod -a
sudo modprobe -v exfat
sudo rm -v /home/ark/exfat.ko*
```

If you’ve altered the /etc/fstab file, be sure that it doesn’t contain a `utf8=1` entry for the exfat partition.  If it does, change it to `iocharset=utf8`

Test and verify that you can still access you exfat partition file after doing the above before you reboot.  If not, something is wrong and you’ll have to reinstall the exfat-fuse package if it can’t be figure out.

As always, backup any important files before doing this just in case something goes horribly wrong.

ridgekuhn, 2022-02-27:
@christianhaitian Awesome, thx! I will try it out and report back when I have some time!

christianhaitian, 2023-04-16:
Closing as there hasn't been an update on this issue for over year.  If it still needs review, just comment here.

closed 2023-04-16 by christianhaitian