christianhaitian/arkos #291: BUG: Diablo 1 no mouse control via right analog thumbstick

Doom0325, 2021-03-14:
I am on the latest version of ArkOS on an ANBERNIC RG351P. Mouse is locked in the right hand corner and cannot move.

christianhaitian, 2021-03-14:
Looks like something with the new kernel caused the mouse to no longer function in Diablo 1.  I can't seem to figure out what's caused the issue as it was the same issue previously experienced on the 64 bit version of Diablo 1.  I've submitted the issue for review upstream in hopes they can help identify a resolve.  https://github.com/diasurgical/devilutionX/issues/943

Rh0gii, 2021-03-22:
for now... should we try to play this off of the 32bit version of Retroarch?

christianhaitian, 2021-03-22:
No solution at this point.

southoz, 2021-04-15:
Mouses!

The only time the mouse location values were updated was when an SDL_MOUSEMOTION event was being triggered and the right stick movement was definitely not triggering that event. 

Update Mouse location within the HandleRightStickMotion routine has fixed the issue on the OGS.

https://github.com/southoz/devilutionX/commit/6a82fdee333bfd68105344914a629caae6e90f8e

christianhaitian, 2021-04-16:
> Mouses!
> 
> The only time the mouse location values were updated was when an SDL_MOUSEMOTION event was being triggered and the right stick movement was definitely not triggering that event.
> 
> Update Mouse location within the HandleRightStickMotion routine has fixed the issue on the OGS.
> 
> [southoz/devilutionX@6a82fde](https://github.com/southoz/devilutionX/commit/6a82fdee333bfd68105344914a629caae6e90f8e)

Nice.  I'll check this out asap and if good, push an update to resolve!

christianhaitian, 2021-04-16:
I've confirmed that this does fix the mouse issue.  This fix will be included in the next update soon.

Rh0gii, 2021-04-16:
is there a manual fix for those that are on say... Retro OZ on OGS...?

Rh0gii, 2021-04-16:
Thanks to everyone that's never given up and found a fix by the way!

christianhaitian, 2021-04-16:
If your comfortable with ssh or winscp, you can just take the executable out of the attached compressed file and put into the /home/ark/.config/devilution folder.  RertroOZ may have a similar setup as well to allow this.  You may have to set the permissions for the file to be executable so it can run.  `chmod 755 devilutionx`

[devilutionx.zip](https://github.com/christianhaitian/arkos/files/6327443/devilutionx.zip)

Rh0gii, 2021-04-16:
I'm familiar with ssh and winscp. I'll give it a shot.  THANKS!

Rh0gii, 2021-04-16:
jumped into my winscp.  there is a ".config" folder in there, but no "devilution" folder within that.  All well.  I'll probably need help from the developer as to where the config folder is hiding for this port.  thanks for the help anyway

southoz, 2021-04-16:
The folder is /home/ark/.config/devilution

```
ark@rg351p:~/.config/devilution$ l -a -l
total 2632
drwxrwxrwx  2 ark  ark     4096 Dec 16 05:35 ./
drwxrwxrwx 15 root root    4096 Apr 16 21:44 ../
-rwxrwxrwx  1 ark  ark  2684440 Dec 16 05:35 devilutionx*
ark@rg351p:~/.config/devilution$
```

Rh0gii, 2021-04-16:
Yeah... I haven't gotten to the point to work on my RG351 running ArkOS.  At the moment, I was trying to get the mouse working for my OGS running RetroOZ.   So... for RetroOZ... I just need to place the executable "devilutionx" file in the /roms/ports/devilution folder and execute it in an ssh session....?  Or should I just stop messing with it before I hurt myself?   (please pardon my n00bness, sometimes, when I try really hard, I know enough to be dangerous - XD )

southoz, 2021-04-17:
RetroOZ 0.42 update includes the updated OGS binary. Keep messing, just backup your roms so you can reimage an SDCard or even better use a second card to mess around with. Breaking things is the first step to fixing them and if you need to ask questions or need help feel free to hit me up in the Odroid Forums. The more people that are tinkering the quicker issues are found and fixes can be made. You know your on track when you break things that bad have to open a serial console to your retro handheld :P

Rh0gii, 2021-04-17:
WOOOO!!!, OK, I now have mouse-love working on my ArkOS powered RG351P!
Thanks to you both for all of your work, help, and patients with me.  As
for my OGS running RetroOZ, yeah, I'm totally down to flashing 0.42 on a
spare microsd and see what I can break!

Again, Cheers to you both!
Rh0gii

christianhaitian, 2021-04-17:
Thanks for confirming that this resolved the issue.  Big thanks to @southoz for the fix.

closed 2021-04-17 by christianhaitian

howdoicomputer, 2021-04-25:
Super excited. I'm patiently waiting for my 351M and now I can use the final build of ArkOS because I'm super eager to relive Diablo before Resurrected comes out.