christianhaitian/arkos #292: Half-Life cannot go up or down ladders

vimstar, 2021-03-14:
I have played Half-Life up to the point where I need to climb up a ladder to continue and I am unable to. When you approach the ladder and try to climb it does nothing no matter what I try. I am playing on the RG 351P using the latest release of ArkOS 1.6 (03082021).

christianhaitian, 2021-03-14:
Try the alternate control configuration described at the end of this section and see if that helps.  https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information

christianhaitian, 2021-03-15:
Hi,  

Have you tried what I suggested already?  Did it help?

Thanks,

vimstar, 2021-03-17:
Sorry about the delayed reply. Yes, I did try the alternate control configuration and it didnt help. I am still unable to climb up or down ladders.

Rh0gii, 2021-03-22:
I'm having the same issue.  in a normal setting, would the "move forward" / "move backward" be used to climb "up" and "Down" respectively?  that's not working in my case.

Rh0gii, 2021-03-23:
I don't know if this is helpful, but FWIW, I found that when I do a fresh  load up of Half-Life and load a saved game by a ladder, I can intermittently use the ladder, but after a few minutes (maybe 1 or 2) it will stop working and just make me "stick" to the ladder where all that I can do is just jump back away from holding the ladder.

Rh0gii, 2021-04-19:
is this an intermittent issue that only some users have? Is the problem something that can be observed? Any ideas for how those of us affected can get this working?

brunns, 2021-04-19:
I'm going to try setting HL up again from scratch, making a note of **exactly** what I did here. The problem might will be the result of some particular sequence of set up steps.

brunns, 2021-04-19:
Can confirm that after following the instructions [here](https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#half-life-1), with or without installing the alternative control set, I **can't** climb the ladder.

For this test, I **didn't** invert the y-axis controls as per the [retrogamecorps instructions](https://retrogamecorps.com/2020/12/23/guide-half-life-on-the-rg351p/#Install) - but I had done that the first time I installed HL, and I did have the stairs issue with that too.

I've upgraded to the latest ArkOS, and I'm using the `valve` data from the Steam version of Half-Life on a Mac. (I used the Mac for all the file copying and so forth, just in case *that* might be the issue.)

christianhaitian, 2021-05-02:
As development for the RG351P has ended, I will not further investigate this issue.  Perhaps krishenriksen may be able to further review and assist on this issue since he was the one who delivered this port.  He also has a version available through AnberPorts.  https://github.com/krishenriksen/AnberPorts

closed 2021-05-02 by christianhaitian