christianhaitian/arkos #293: RetroArch 64bit no text issue

RiseAgainst88, 2021-03-15:
Hey christianhaitian,

I am using the latest firmware update "v1.6 (03/08/2021)"
Since then I have a problem with the 64bit version of RetroArch (32bit version runs well), I can`t see any text.
the background is moving and it seems that everything else is going right.
also, if I want to save or load my progress ingame, the save/load indicator is colorful and without text.

[image]
[image]

Thanks in advance and thank you for this great system!

duskgao, 2021-03-15:
set your retroarch's user language to english maybe help

RiseAgainst88, 2021-03-15:
it is already set to english, didnt change the language but thanks for the idea.

christianhaitian, 2021-03-15:
What happens when you load retroarch 64 bit by itself?

RiseAgainst88, 2021-03-15:
The same

christianhaitian, 2021-03-15:
I've been trying to replicate the issue without success.  I suspect this is due to some custom setting you may have set such as choosing xmb for the menu driver instead of just using rgui but I'm not sure that may not work with the new kernel.  Again, I can't replicate the issue.  Let's try restoring your retroarch 64 bit setup back to the stock configuration.  Restore the following backup using the steps below and see if that resolves the issue for you.

1.  Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2.  Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3.  Once completed, reboot the device then test Retroarch 64 bit.

Hopefully this resolves the issue for you.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6144705/arkosbackup.tar.gz)

RiseAgainst88, 2021-03-15:
Yes! 
Now its working fine for what I can tell. 
Strange Issue, for sure. 
Thanks for your fast help and keep up the good work! 

[image]

christianhaitian, 2021-03-15:
Great.  It must have been some setting that's not compatible with the kernel.  Enjoy!

closed 2021-03-15 by christianhaitian