christianhaitian/arkos #294: Feature request: Adding Love2d game engine

ludimar75, 2021-03-15:
Hi christianhaitian , I would like to add Love2d as feature request. Is a 2d game engine better than lutro and chailove with a lot of games development on https://itch.io/games/tag-love2d

That's Love retropie docs : https://retropie.org.uk/docs/Love/ and that's the source on github: https://github.com/love2d/love

thanks a lot for your hard working

ludimar75, 2021-03-15:
Great thanks

trelemar, 2021-09-19:
I second this. I installed love and i can get it to run but it closes after trying to create a window. Not really sure what to do to get it to work. "Failed to create EGL surface"

konsumer, 2022-05-21:
It seems like headers that should be in /usr/include are missing.

```
dpkg -L libsdl2-dev
# ... a bunch of files
/usr/include/SDL2/SDL.h

ls -al /usr/include/SDL2
ls: cannot access '/usr/include/SDL2': No such file or directory
```

sometimes [megasource](https://github.com/love2d/megasource) can help with building love for systems that have screwed up deps, as it includes locked-in versions of all the deps (it was designed to make it easier on windows, where it can be a pain to collect all this.)

```
git clone https://github.com/love2d/megasource.git
cd megasource
cmake -B build --target megatest --config Release .

#... mostly goes ok, then this:

CMake Error at libs/SDL2/cmake/macros.cmake:36 (message):
  *** ERROR: Missing Xext.h, maybe you need to install the libxext-dev
  package?
Call Stack (most recent call first):
  libs/SDL2/cmake/sdlchecks.cmake:431 (message_error)
  libs/SDL2/CMakeLists.txt:1267 (CheckX11)
```

same prob:

```
dpkg -L libxext-dev | grep Xext.h
/usr/include/X11/extensions/Xext.h

ls -al /usr/include/X11/extensions/Xext.h
ls: cannot access '/usr/include/X11/extensions/Xext.h': No such file or directory
```

Did I screw up my install of ArkOS or have all the important headers from `-dev` packaged been deleted without uninstalling them?

konsumer, 2022-05-21:
I did this to make sure the headers were in place:

```
sudo apt-get --reinstall install libxext-dev

ls -al /usr/include/X11/extensions/Xext.h
-rw-r--r-- 1 root root 1655 Aug 24  2019 /usr/include/X11/extensions/Xext.h
```
 but it's still not finding it. Wondering if something else is broken with include-paths?

```
-- Looking for X11/Xcursor/Xcursor.h
-- Looking for X11/Xcursor/Xcursor.h - not found
-- Looking for X11/extensions/Xinerama.h
-- Looking for X11/extensions/Xinerama.h - not found
-- Looking for X11/extensions/XInput2.h
-- Looking for X11/extensions/XInput2.h - not found
-- Looking for X11/extensions/Xrandr.h
-- Looking for X11/extensions/Xrandr.h - not found
-- Looking for X11/extensions/Xfixes.h
-- Looking for X11/extensions/Xfixes.h - not found
-- Looking for X11/extensions/Xrender.h
-- Looking for X11/extensions/Xrender.h - not found
-- Looking for X11/extensions/scrnsaver.h
-- Looking for X11/extensions/scrnsaver.h - not found
-- Looking for X11/extensions/shape.h
-- Looking for X11/extensions/shape.h - not found
-- Looking for include files X11/Xlib.h, X11/extensions/Xdbe.h
-- Looking for include files X11/Xlib.h, X11/extensions/Xdbe.h - not found
-- Looking for include files X11/Xlib.h, X11/extensions/xf86vmode.h
-- Looking for include files X11/Xlib.h, X11/extensions/xf86vmode.h - not found
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xext.h
-- Looking for 3 include files X11/Xlib.h, ..., X11/extensions/Xext.h - not found
CMake Error at libs/SDL2/cmake/macros.cmake:36 (message):
  *** ERROR: Missing Xext.h, maybe you need to install the libxext-dev
  package?
Call Stack (most recent call first):
  libs/SDL2/cmake/sdlchecks.cmake:431 (message_error)
  libs/SDL2/CMakeLists.txt:1267 (CheckX11)
```

konsumer, 2022-05-22:
I hadn't seen [this](https://github.com/christianhaitian/arkos/tree/main/Headers).

So I did this:

```
wget https://github.com/christianhaitian/arkos/raw/main/Headers/install_headers.sh
chmod 777 install_headers.sh
./install_headers.sh
```

That didn't seem to make a difference (the files are still missing and things don't build.)

I reinstalled all installed packages like this:

```
sudo apt install aptitude
sudo aptitude reinstall '~i'
```

And I get this error:

```
E: Can't find a source to download version '1.7-2+deb10' of 'libmali-rk-bifrost-g31-rxp0-wayland-gbm:arm64'
E: Can't find a source to download version '2.0.10+dfsg1-1ubuntu1' of 'libsdl2-2.0-0:arm64'
E: Can't find a source to download version '4.4.189-2' of 'linux-headers-4.4.189:arm64'
E: Can't find a source to download version '4.4.189-139502-g380eeff98d35-4' of 'linux-image-4.4.189-139502-g380eeff98d35:arm64'
E: Can't find a source to download version '2.0-1' of 'retroroller:arm64'
E: Can't find a source to download version '1.0-1' of 'sleepkernel:arm64'
E: Can't find a source to download version '1.0.0-1' of 'emulationstation-go2:arm64'
E: Can't find a source to download version '1.0.0-1' of 'emulators-32bit-go2:armhf'
E: Can't find a source to download version '1.0.0-1' of 'emulators-64bit-go2:arm64'
E: Can't find a source to download version '20200224-1' of 'libgo2:arm64'
E: Can't find a source to download version '20200224-1' of 'libgo2:armhf'
E: Can't find a source to download version '1.7-1' of 'libmali-gbm:armhf'
E: Can't find a source to download version '1.7-1' of 'libmali-gbm:arm64'
E: Can't find a source to download version '20191031-1' of 'librga:armhf'
E: Can't find a source to download version '20191031-1' of 'librga:arm64'
E: Can't find a source to download version '4.4.189-14' of 'linux-odroid-go2:arm64'
E: Can't find a source to download version '20200224-1' of 'retrorun-go2:armhf'
E: Can't find a source to download version '20200224-1' of 'retrorun-go2:arm64'
E: Can't find a source to download version '1.0.0-2' of 'utils-go2:arm64'
After unpacking 0 B will be used.
E: Can't find a source to download version '1.7-2+deb10' of 'libmali-rk-bifrost-g31-rxp0-wayland-gbm:arm64'
E: Can't find a source to download version '2.0.10+dfsg1-1ubuntu1' of 'libsdl2-2.0-0:arm64'
E: Can't find a source to download version '4.4.189-2' of 'linux-headers-4.4.189:arm64'
E: Can't find a source to download version '4.4.189-139502-g380eeff98d35-4' of 'linux-image-4.4.189-139502-g380eeff98d35:arm64'
E: Can't find a source to download version '2.0-1' of 'retroroller:arm64'
E: Can't find a source to download version '1.0-1' of 'sleepkernel:arm64'
E: Can't find a source to download version '1.0.0-1' of 'emulationstation-go2:arm64'
E: Can't find a source to download version '1.0.0-1' of 'emulators-32bit-go2:armhf'
E: Can't find a source to download version '1.0.0-1' of 'emulators-64bit-go2:arm64'
E: Can't find a source to download version '20200224-1' of 'libgo2:arm64'
E: Can't find a source to download version '20200224-1' of 'libgo2:armhf'
E: Can't find a source to download version '1.7-1' of 'libmali-gbm:armhf'
E: Can't find a source to download version '1.7-1' of 'libmali-gbm:arm64'
E: Can't find a source to download version '20191031-1' of 'librga:armhf'
E: Can't find a source to download version '20191031-1' of 'librga:arm64'
E: Can't find a source to download version '4.4.189-14' of 'linux-odroid-go2:arm64'
E: Can't find a source to download version '20200224-1' of 'retrorun-go2:armhf'
E: Can't find a source to download version '20200224-1' of 'retrorun-go2:arm64'
E: Can't find a source to download version '1.0.0-2' of 'utils-go2:arm64'
E: Internal error: couldn't generate list of packages to download
E: Perhaps the package lists are out of date, please try 'aptitude update' (or equivalent); otherwise some packages or versions are not available from the current repository sources
```

I think I am going to try to get a minimal debian working in docker with correct arch. Not sure how else to proceed with this, as it seems very broken.

konsumer, 2022-05-22:
Well, hmm. So I still can't build it, but I ran `sudo apt install love` and then `love` in ssh, and it appears to run ok.

I put 2 games in /roms/love (mari0.love and pong.love, both tested on desktop) and added this to  /etc/emulationstation/es_systems.cfg:

```xml
<system>
        <name>love</name>
        <fullname>Love</fullname>
        <path>/roms/love</path>
        <extension>.love</extension>
        <command>love %ROM%</command>
</system>
```

Restart emulationstation, and it seems to work fine, at least graphically. I'd need to do more testing, but it seems to be correctly using acceleration on framebuffer, and stuff.

christianhaitian, 2022-05-22:
Hello,
You should've used the vm I have available for builds like this. [https://forum.odroid.com/viewtopic.php?p=306185#p306185](https://forum.odroid.com/viewtopic.php?p=306185#p306185)

Thanks for testing.  How are controls?

konsumer, 2022-05-22:
> You should've used the vm I have available for builds like this.

That seems great! I will do this in the future. I work on the node-raylib team, and tend to do all the "lowend devices" testing, and love seems to run a bit smoother on this than GPi case with pizero2w!

> How are controls?

Seems pretty good. In the above demo it basically just plays music & tests controls onscreen (which can be helpful for figuring out which buttons are which for love games.) You can view the source by extracting the zip file and looking at main.lua.

In your game, make sure to make an exit function like I did with start+select, because it doesn't have that by default, and you want people to be able to exit without restarting.

I am working on a theme page for epicnoir right now for `love` & `apps`, so it looks nicer. Should I PR these changes?

christianhaitian, 2022-05-22:
Which git are you updating for epicnoir?  The one used in ThemeMaster.  If so, it would be helpful to the maintainer if you did a PR I think.

konsumer, 2022-05-23:
> Which git are you updating for epicnoir? The one used in ThemeMaster. If so, it would be helpful to the maintainer if you did a PR I think.

I am just editing the the one on my SD card, directly, but I am happy to PR to wherever.
[image]

konsumer, 2022-05-23:
It seems like the update goes kind of deep: 
- JohnIrvine1433/ThemeMaster
- Jetup13/Emulationstation-OGA-Theme-Gallery
- Jetup13/es-theme-epicnoir
- c64-dev/es-theme-epicnoir

And a couple are missing issue queues. I guess I could just PR to the last one, but it doesn't seem like it will end up in the Jetup13 repos.

Here is the changes I made, if anyone wants them:
[es-theme-epicnoir.zip](https://github.com/christianhaitian/arkos/files/8750562/es-theme-epicnoir.zip)

konsumer, 2022-05-23:
I made one for `apps` too, since it's useful to me to be able to run any random thing from the menu (so I don't need a keyboard connected to device to test.)

[image]

First I added this to `/etc/emulationstation/es_systems.cfg`:

```xml
<system>
        <name>apps</name>
        <fullname>Applications</fullname>
        <path>/roms/apps</path>
        <extension>.sh</extension>
        <command>%ROM%</command>
</system>
```

Then made these changes to theme folder:

[es-theme-epicnoir.zip](https://github.com/christianhaitian/arkos/files/8750610/es-theme-epicnoir.zip)

christianhaitian, 2022-11-19:
This has been accommodated with a previous OTA Update.

closed 2022-11-19 by christianhaitian