christianhaitian/arkos #299: BBC or Acorn Electron Emulation?

SuperBadger77, 2021-03-19:
Hi,

This is a request for a new emulator. Could I please request the addition of a BBC Micro emulator and Acorn Electron? Had many a fond day playing on these as a child - they got me into computing. I know they are largely keyboard based, but are similar to the Amstrad 464 and C64 - and some of the games are iconic, such as Elite. There are several versions of a BBC Emulator - the main one being Beeb-em which is available on unix \ windows \ mac. However, what really got me interested in this is that someone ported it to the RG350/P/M - and updated the code to provide default key mappings, and the easy ability to map ked to keys in app for each game.

Link to the Beebem sites for source code, versions and supported ports:
http://www.mkw.me.uk/beebem/
http://beebem-unix.bbcmicro.com/

Forum where the the RG350 version was released with key mapping for each game: https://stardot.org.uk/forums/viewtopic.php?t=18518
Github: https://github.com/jeremyrayner/beebem-rg350/
Github for various Beemem projects and systems.

If you take the code from the RS97 or the RG350, make it run on the rk3326 with the ability to configure joypad to key mappings for each game, then this would open an amazing set of games to the chipset and various platform.

Thank you

Rik

christianhaitian, 2021-04-27:
I'm still reviewing the possibility of getting a decent working version of this emulator going.  The issue is getting it to build in my dev environment.  I'm going to try adding sdl1.2 and see if it makes a difference.

SuperBadger77, 2021-04-27:
Great - thank you for looking into it - I really appreciate your efforts and all the work you have done - as do many others. If you get it working: great, if you don't; at least you tried :-)

christianhaitian, 2021-04-30:
Sorry as I was not able to get a BBC Emulator going for ArkOS.  I wish you well.

closed 2021-04-30 by christianhaitian