christianhaitian/arkos #301: Keys (including global hotkeys) disabled upon return from safe sleep mode on retrorun flycast_rumble and flycast_extreme core for Sega Dreamcast

fry-guy, 2021-03-21:
### Describe the Issue (If applicable)

Controls are deactivated upon return from safe sleep mode (tapping the power button) in the retrorun32 flycast_rumble and flycast_extreme cores for Sega Dreamcast. This includes the global hotkeys start+select, so not only you not return to playing the game, you need to hard rest the device to use it again at all.

### How can the issue be reproduced? (If applicable)

This has been occurring while playing Skies of Arcadia. I have not tested it other Dreamcast games.

### What device are you using?

RG351P

### What version of ArkOS are you on?

 v1.6.03.20.2021

### Anything else you'd like to include that may help to review this issue or feature request?

Thank you again for your incredible work on ArkOS!

christianhaitian, 2021-03-21:
Unfortunately, it's a common issue with sleep on this device.  You can still do R3+Power to shutdown safely when this issue occurs.  I've added this information tot he known issues list on the wiki.

closed 2021-03-21 by christianhaitian

fry-guy, 2021-03-21:
Thank you! I did not know about the same shutdown. Is the a full list of hotkey combos anywhere? Many seem similar to the ones listed here: https://forum.odroid.com/viewtopic.php?p=284237&sid=d77c38389083c2698752b3525e4d5ddc#p284237

I assume this core doesn't do savestates? Any chance they will be added?

christianhaitian, 2021-03-22:
Yes.  All of that information is available in the FAQ section of the wiki.  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS.  I'm hoping to get savestates working for retrorun eventually.