christianhaitian/arkos #304: Hotkeys doesn't work on RG351P

Captain69Wolf, 2021-03-27:
Hi There, 
I see that there is a problem when you quit retroarch menu in-game by pushing X + select. But I can't fix it. I change the hotkeys in both retroarch menu with L1 + R1 + Start + Select but the problem is that Start + Select immediatly quit the game, so I take L3 + R3 but it doesn't work, the only hotkey that still work is Start + Select witch quit the game. I try to reset to defaut the option in both retroarch, but it doesn't work.
I try to format my SD card and start from scratch but I still have the same problem. I even try the disable the fullscreen, I see that it helped for a similar problem on windows.
I know it must be a stupid problem with a simple solution but I really don't know what to do.
Can someone help me for fixing this issue?

christianhaitian, 2021-03-27:
Go to Settings > Input > Hotkey Binds > Press Quit Twice is On.

Captain69Wolf, 2021-03-27:
It's  already ON on both retroarch settings, the thing is it's still stop the game the first time I push select + start.

christianhaitian, 2021-03-27:
I'm not sure why you're seeing settings in French but it sounds like many things have been changed in the settings.  Do you have another SD card you can install ArkOS on and test if everything works as it should before you start making changes?

Captain69Wolf, 2021-03-27:
Settings are in French because I am French^^ Unfortunately I have no SD card but I format this one and re-install arkOS on it and the issue remain. So it doesn't bother to making changes, there is no whorse case scenario than the lake of menu in-game to scalling and tweaking the games^^ If nothing work I will buy a fresh new sd card, but I am not even sure it fix this problem. And I only change a lot of settings in hotkeys menu, after the issue. I even rest to default (I don't have many saves already in my device, so I don't care to reset from scratch)

christianhaitian, 2021-03-27:
In order for me to to help further, I would have to be able to replicate the issue and I can't replicate this one.

Captain69Wolf, 2021-03-27:
Ouch, well I will get a new sd card and install arkOS on it and hope it fixes the issue. Thanks you very much for taking the time to help me.

Captain69Wolf, 2021-03-28:
There is update! I realise that the issue comes with retrorun 32 (I had only Saturn and Dreamcast games in my Arkos SD cards). I try with a different core and the hotkeys works... So I guess I had to go into the retrorun option, but I don't know how to do it.

christianhaitian, 2021-03-28:
I see.  Retrorun and retrorun32 do not allow button changes or other customizations like retroarch does.  Those settings are hardcoded into the retrorun program.  It's one of the drawbacks of using that emulator.  For pause, the indication will be that the emulation stops until you hit the combination again and then it starts again.  For Fast Forward emulation, you'll just see the emulation moves a little quicker.  If you're familiar with programing, you can recompile retrorun  using the sources [here](https://github.com/christianhaitian/retrorun-go2).

closed 2021-03-28 by christianhaitian