christianhaitian/arkos #305: batt_life_warning.py fails on OGA 1.0.

thomas001, 2021-03-27:
### Describe the Issue (If applicable)

`/usr/local/bin/batt_life_warning.py` fails to open the GPIO device and thus fails. 

### How can the issue be reproduced? (If applicable)
```
$ /usr/local/bin/batt_life_warning.py
Traceback (most recent call last):
  File "/usr/local/bin/batt_life_warning.py", line 30, in <module>
    if int(open(pwr_led, "r").read()) == 1:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/gpio/gpio77/value'
```

### What device are you using?
OGA 1.0

### What version of ArkOS are you on?
03202021

christianhaitian, 2021-03-27:
That file is no longer used.  ogage has replaced it to provide hotkey control.

thomas001, 2021-03-27:
Oh, thanks for the update. So is the file just a leftover from a previous version that was not removed during an update?
It seems to be spawned by systemd.

christianhaitian, 2021-03-27:
Oops, sorry I misspoke.  That file is valid.  I may need to issue an update to correct that.

christianhaitian, 2021-03-27:
Hi,

Until I can release an update to resolve this, you can restore the attached backup that should fix that python script issues.  

1. Take the linked compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, reboot the system and the python script should be resolved and working as it should be which is:
    -  Screen brightness will lower then raise to previous set brightness every 30 seconds when battery life is less than 30 percent
    -  Screen brightness will lower then raise to previous set brightness every 5 seconds when battery life is less than 19 percent

Please confirm if it resolves the issue for you or not.  

Thanks for reporting this.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/6216384/arkosbackup.tar.gz)

thomas001, 2021-03-29:
Thanks for the quick fix.

I updated the files from your backup and the error message is gone. The service is marked as successfully started and running by systemd. My battery level has yet to drop, I will post an update if the behavior is correct one that happens.

thomas001, 2021-03-30:
The screen flashing feature works here. It already started flashing every 5s when the battery indicator showed "20", but that's probably some rounding going on when displaying it.

Thanks again for the swift fix.

closed 2021-03-30 by thomas001