christianhaitian/arkos #306: Green blinking cursor

notimp, 2021-03-27:
I have a quick question. I'm in the process of mitigating over a large retroarch installation from another SD card, and am touching basically a bunch of files on the sdcard of ArkOS. Somewhere during that process, the cursor in ArkOS (the one you see blinking during 'load screens'), started to be drawn in green instead of white.

Why is that - and how can I prevent it. It triggers my OCD - so I'm afraid I had to ask, and I'll do almost anything not to have to see it. ;)

Two additional short questions, retroarch 32 is only set up for a few emulators, correct? Which are those (havent looked into the respective core folders yet, probably should) and which installation of retroarch would you default to for most systems?

Also - how would I go about trying to launch (one of the versions) on boot. I'd rather navigate retroarch than ES, if I can choose.

Sorry, to have made this somewhat of a 'support request thread' -- but If only I can get rid of the green cursor - the rest I probably could figure out somehow.. :)

christianhaitian, 2021-03-27:
That may be caused by the enabling remote services feature if you're using that regularly.  You can putting `printf "\033[0m"` in /usr/local/bin/perfmax  just below the `#/bin/bash` line.  

retroarch32 is primarily there for pcsx_rearmed which has the best performance for psx emulation.  I believe the 32bit version of the atari lynx emulator is the default emulator as well.

Launching into retroarch directly instead of ES is a more involved process that I don't recommend for this distribution as many of it's functions are designed around ES being the frontend.

closed 2021-03-27 by christianhaitian

notimp, 2021-04-02:
> That may be caused by the enabling remote services feature if you're using that regularly. You can putting `printf "\033[0m"` in /usr/local/bin/perfmax just below the `#/bin/bash` line.
>
> retroarch32 [...]

Thank you _very_ much.