christianhaitian/arkos #308: Can't access PPSSPP menus with ArkOS 1.6 on RGB10

RodrigoReisBorn, 2021-03-30:
Hello,

I installed de RGB10 rom and executed de update script.
After that I've put some PSP roms and tried to play. But I can't access the ppsspp menus and, more than that, tha R and L buttons don't seem to work.

Thanks for your efforts and work.

christianhaitian, 2021-03-30:
It sounds like you deleted the original folder contents of the psp folder.  Try restoring the contents from the link below and see if that fixes the issue for you. https://github.com/christianhaitian/arkos/raw/main/defaultromsfolderstructure-rgb10.7z

RodrigoReisBorn, 2021-03-30:
Perfect! That's exactly what I was doing, because I was copying tha psp games folder from the other rom I was using.

Thanks a lot!

closed 2021-03-30 by RodrigoReisBorn