christianhaitian/arkos #309: Feature Request: RG351V Support

mikehamil10, 2021-03-31:
The stock firmware sucks, it would be great to get ArkOS on my RG351V!!!

damien6669, 2021-03-31:
https://github.com/christianhaitian/arkos/issues/300

christianhaitian, 2021-03-31:
I do have ArkOS running on the 351v, however, Anbernic has promised to release sources so I am waiting for that to see if using them can improve the experience with ArkOS on the 351v.  Right now, it has annoying issue where on initial boot, it takes about 10-12 seconds for the vibration to stop.  Very annoying!  I'm hoping with their sources, we simply turn off vibration on boot period!  If not, at least shorten it to just a few seconds like on the 351p/m.  Stay tuned.

closed 2021-03-31 by christianhaitian

nullstalgia, 2021-04-01:
I would recommend pinning one of these to the top to avoid more being made :)

ajsaucier, 2021-04-02:
Speaking of the vibration on a 351P when it boots up, is it possible to turn that off completely? I haven't found anything about that in the wiki.

NeoTheOne, 2021-04-02:
> Speaking of the vibration on a 351P when it boots up, is it possible to turn that off completely? I haven't found anything about that in the wiki.

Don't worry my guy it's done by the OS and not the device itself at boot.

ajsaucier, 2021-04-02:
> > Speaking of the vibration on a 351P when it boots up, is it possible to turn that off completely? I haven't found anything about that in the wiki.
> 
> Don't worry my guy it's done by the OS and not the device itself at boot.

Does that mean it can be turned off somehow?

nullstalgia, 2021-04-02:
We're waiting for Anbernic to release the sources for the bootloader to
tweak the vibration and other things. So, just sit tight for now. :)

christianhaitian, 2021-04-02:
I haven’t found a way to completely turn off vibration on boot yet.

nullstalgia, 2021-04-03:
Oh nice, sources are out.

Just so this can be here for others looking for it:

https://github.com/SummerSunGenius/RG351V_kernel

https://github.com/SummerSunGenius/RG351V_buildroot

https://github.com/SummerSunGenius/RG351V_uboot

christianhaitian, 2021-04-07:
Vibration is down to about 2 seconds on initial boot.  Shouldn’t be to much longer before ArkOS is out.

markb44z, 2021-04-07:
This is such great news, been waiting for this ever since I got my RG351V, ordered a brand new 16GB SanDisk SD card in preparation, can't wait!

christianhaitian, 2021-04-17:
Test release available via one of the links below.  Be sure to do an online update after install for the latest and greatest improvements. 

Google Drive: https://drive.google.com/file/d/1AP5VFPN11LWnryvQS68Hpy3zkIn3jIpr/view?usp=sharing
Mega: https://mega.nz/file/SVoRRAzD#SJ5kS9B7nfPZQ4M_DgR0TjOozPrbdztqhDA-V5FpbBQ
FAQ: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V

*Notes* 
-  For NDS (Drastic emulator) - Do not try to remap the stylus up, down, left and right buttons.  Doing so will cause the stylus movement to be erratic or not function at all.
-  Hotkey information: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS
-  See this FAQ link for information on how to make use of the slot 2 for your games and roms.  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games

nullstalgia, 2021-05-01:
Is this image still the latest available? When I try to update it after flashing to the card and enabling remote services, it hangs in the middle of the update. About to test 351ELEC to see if it's just a bad SD card.

christianhaitian, 2021-05-01:
It's still available and being updated regularly.

notimp, 2021-05-02:
Just got it up and running today. :) As of now all games (even GBA) are stretched to 4:3 and the aspect ratio setting in retroarch doesnt work. This is my Nr.1 complaint. If you could address this, everything else pretty much works quite well.

I've one more quarrel with the system as of now, and that is, that the displays colors in my case are just off my enough to bother me. Especially on DS games (tested Phoenix Wright) they have a brownish tint to them, and maybe look a tad desaturated. I understand that this most likely is the screen used itself, but just in case you might be able to look into it - I thought I'd mention it as well.

(Color accuracy on the first gen RK2020's was way better, btw. :) )

Oh, and thank you for allowing users to switch the led color in the middle of the device to red, its much more mute and less distracting. Thank you for the lateral thinking on this one. :)

christianhaitian, 2021-05-02:
The screen is 4:3 so no surprise with everything being 4:3.  GBA is probably the most impacted and there's not too much you can do to fix that but try different aspect ratios to see what works best for you.  You'll need to turn on Integer Scale then you can try the different Aspect Ratios depending on what you prefer.  

I have no qualms or issues with the screen on my end.  The colors are very vivid and the resolution can't be beat for the options available for this chips.  Perhaps you unit has a fault with the screen as I haven't heard much of issues with the screen.  

The RK2020 V1 has nothing on this unit other than being more pocketable but that's my opinion. :)

notimp, 2021-05-02:
Thank you for the reply - even with integer scale on, at least in the mgba core - the core provided aspect ratio is off, 3:2 aspect ratio is off (provides a square'ish aspect ratio), and custom isnt fine grained enough you could set a default gba aspect ratio.

So - the entire aspect ratio behavior currently is off. (Normall aspect ratios also would take without having to turn on Integer scale. Also their behavior is not what would be expected in terms of actual aspect ratio behavior.

Probably a pain to fix - but please take another look if you can.

Also it might just be me, but the brownish tint seems to be fading... so maybe (fingers crossed) in my case it actually was a 'glue' layer in the screen assembly thats hardening out and getting transparent. Something is off regardless, contrast is lacking at the top end and gamma looks like it crushes darker colors (bad gamma curve?) - probably luck of the draw and the screen though.

christianhaitian, 2021-05-02:
I don't think this is possible.  Have a look at retrogamecorps video on the 351v about this.  It's what it is with this device.  To be sure. Try stock or 351elec and see if the results are any different for you there.  If not, you just need a 3:2 device for your gba emulation needs.  The rk2020 spoils you there.  :)

christianhaitian, 2021-05-03:
**ArkOS Test Release 2.1 is now available**

Google Drive: https://drive.google.com/file/d/1xRp8adMwWDCs_JqQJWd6sef9t5Bddzjd/view?usp=sharing
Mega: https://mega.nz/file/2Qo2kC5J#JosOCBh9CWRy4WEHVz2Cg8O5VkPoonSI-BgQiZ3dZWk
FAQ: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V
MD5: 51713AC3563FEBDAADABE09CAC141150

**Important Notes** 
- Follow the instructions from the wiki to properly loading this image: https://github.com/christianhaitian/arkos/wiki#instructions-for-loading
- Don't use the original micro sd card that came with the unit.  Those are not good cards.  Use authentic Samsung or Sandisk micro sd cards for the best reliability and performance.
-  For NDS (Drastic emulator) - Do not try to remap the stylus up, down, left and right buttons.  Doing so will cause the stylus movement to be erratic or not function at all.
-  Hotkey information: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-what-are-the-global-event-keys-and-emulator-event-keys-in-ArkOS
-  See this FAQ link for information on how to make use of the slot 2 for your games and roms.  https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-how-do-i-make-use-of-the-second-sd-slot-for-roms-and-games
- Changelog: https://github.com/christianhaitian/arkos/blob/main/changelogs/rg351v-changelog

Make sure to Update online via the Options section for the latest fixes and features as they become available.

notimp, 2021-05-05:
> I don't think this is possible. Have a look at retrogamecorps video on the 351v about this. It's what it is with this device. To be sure. Try stock or 351elec and see if the results are any [...]

I've now 'answers' on all my issues. :) So just for clarification: The Retrocorps video ( https://www.youtube.com/watch?v=ZzpjIdpZPRA ) shows 3:2 scaling being set correctly in retroarch on the default (emuelec based) firmware, without integer scaling having to be enabled. This solves the GBA aspect ratio issue. On ArkOS this is currently not possible (updated to the latest possible firmware), as all non-integerscaling based scaling is "broken" in the firmware. The workaround to use integer scaling with GBA games is not really workable - huge black bars on all sides and the image becomes too small.
-

What I perceived as a "gamma issue" actually was RGA scaling being set to enabled on a SNES game. Not recommended. It fills up the image content with extra white pixels at the edges, thereby causing a look as if the screens gamma setting was way off. Easy fix - turn RGA scaling off. (Maybe also by default for the RF351V. :) )

The brown screen tint is still an issue and sadly not imagined - the retrogamecorps shows this as well. This is probably the hardest to 'fix' as knowledge about color science, and driver/kernel level skills would be required, so nothing I really would hope for. But for the record, its not just me, its a general issue. :)

The screen also still has contrast issues (afair red/orange running out at the high end (high saturation colors)), and poor black level, and thats purely hardware - and nothing anyone could fix, short of replacing it. ;)
-

So in order of "severity" - retroarch scaling not "taking" in non-integerscaled mode is a big one, and not an issue in the official firmware. Probably hard to fix, but probably necessary for many people. Certainly for me. (Would be a cause for me to Switch to a different firmware longterm. GBA games are important. ;) )

Being able to counteract the brown tint would be a very nice plus - but at the same time entirely unterstandable if its beyond the scope of the project and not possible.
--

Also - something I noticed -- the flycast_rumble core on my RK2020 (newest model (colored buttons)) in retroarch32 performs _far_ better than any DC emulator (and even flycast_rumble) itself on the RG351V. And actually outperforms all other DC cores, including the retrorun ones.

I've yet to test if this is due to SD card write speed, as part of the performance improvement Is due to the "Enable RTT buffer setting" in emulator, and my current main system SD card is a cheap chinese one, but if its not due to that, or additional system overhead on the RG351V, please take note of the emulation core (flycast_rumble in Retroarch32) that came bundled with the RK2020 1.6 release being _far_ more performant, than any of the DC cores on the current release. (Same settings, everything else tested.) Didnt test replacing (swapping) cores themselves, but will do so in the future. :)

notimp, 2021-05-05:
In terms of color tint - I found a way to compensate. :) Color performance on the RG351V gets better, if you turn the screen brighter, at the same time, doing that with RGA scaling enabled causes those strange shifts that look like shifts in gamma.

If you turn on RGA scaling, and enable the lcd3x filter (in the handheld folder if you aresearching in filters) the entire screen becomes darker, which allows you to push one brightness level higher (at the lower end of the brightness scale), leading to more accurate colors (greens and skintones), the "gamma shift" that RGA scaling produces is beneficial in this case. Probably the best looking image (at lower brightness levels) I got out of the device so far. There is one downside (apart from the LCD filter affecting your image - which you might like or not, but its subtle :) ) and that is raised black level.

Also make sure to put on the screen protector that ships with the device (tempered glass) imho - it lowers overall brightness by a bit (that helps black level ;) ), makes RGA scaling a little less harsh - anything that helps, really. ;)
__

Tried out the TheRA build for a while - which has non-integer scaling working (so perfect GBA image), which looked incredibly good on the RGA351Vs screen. The rest of the build is cluttered and overlayed with effects, and I switched back to ArkOS after about 3 minutes. ;) Couldn bare it.
__

flycast_rumble core, for some reason becomes less performant on 2.3 than on 1.5 - 1.6 ish builds on the device over time - so in the end there is a noticeable difference to the RK2020 on 1.6 - even accounted for SDcard speed - something reduces performance of the core on current test builds (2.3). Probably hard to trouble shoot, but I thought I'd mention it.
__

Duckstation works, probably even on 2x resolution scale on this build, It doesnt on 1.6 on the RK2020, but if you take the binary from the current RG351V build, and swap it with the on on the RK2020, that now can run both duckstation, and flycast_rumble at a speed where Soul Calibur becomes playable and doesnt suffer from frequent stutters. :) (Yay!)
__

Integer scaling "works" on GBA games if you use custom resolution, and then set resolutionscaling for height and with to 3x - then you get a "zoomed in" 3:2 image, with the left and right parts cut off (to 3:2). But the aspect ratio is correct. (This is probably what christianhaitian referred to in the first place - when guiding me towards integer scaling enabled).
__

Please look into getting non-integer based scaling to work, for the sake of being able to play GBA games at their optimal aspect ratio, wile filling the screen. A year from now - doesnt matter, but please give the implementation another shot, and therefore help to make this a true "alrounder". :)

christianhaitian, 2021-05-05:
Very odd as TheRA and ArkOS use the same exact version of Retroarch.  Same binaries as we compiled those and shared them together both slaminger and I.

notimp, 2021-05-05:
> Very odd as TheRA and ArkOS use the same exact version of Retroarch. Same binaries as we compiled those and shared them together both slaminger and I.

The Retroarch build TheRA is using is missing RGA scaling, and afair, wasnt 1.9.1 either (could be wrong on the second part). So there are differences.

christianhaitian, 2021-05-05:
it's 1.9.1.  Like I said, we're both using the same exact binaries.  Not sure on the RGA scaling issue.  I checked in the discord forum I normally hang out in, and no one seems bothered by GBA scaling in ArkOS so far.  They say it's fine and some are very picky about ratios so I don't know on this just yet.  As for flycast32_rumble core, for best dreamcast performance, you have to use the retrorun32 emulator.  The flycast32_rumble core is the exact same core used among all of the ArkOS distributions.  Exactly build.  The only difference is that this device is 640x480 natively while the others are 320x480 which may cause some slower performance on the V since it's the exact same chipset among all devices.  I compensated for that by setting the resolution to default to 320x240 for retrorun emulator use in dreamcast.  To enable 640x480, you have to hold down the a button while launching a dreamcast game using retrorun.

christianhaitian, 2021-05-05:
I also noticed when RGA scaling is turned off, I can change aspect ratios for GBA without turning on integer scale.  Perhaps this is what you're looking for?

christianhaitian, 2021-05-06:
I just released a new update today with Retroarch 1.9.2 with RGA scaling no longer available as an option.  It was causing too much confusion and didn't work right for the V anyway.  Perhaps this will solve the GBA scaling issues you had.

retrogamecorps, 2021-05-06:
Hi, just checked, and yep it was RGA scaling turned ON that was causing the issues @notimp was experiencing.  I was able to replicate the issue and resolve it by turning off RGA scaling and then adjusting the aspect ratio.  In general, I would recommend RGA scaling off across the board -- it's not needed with 640x480 4:3 displays, as far as I can tell.  There is still some pixel distortion when scaling non-integer up to the full display on most systems, but RGA scaling causes a net larger distortion than that natural distortion (which can be fixed with shaders that have better clarity anyway).

notimp, 2021-05-06:
I just woke up, finished a new sdcard setup (exfat partition got corrupted and I had to anyhow. :) ), noticed there was a 2.4 build, installed, and had the non-integer scaling working. :)

Now I miss RGA scaling.. ;) But retroarch guys took it out, so slave to the new default I guess.. - oh well... ;)

Thank you very much for looking into it, and yay for aspect ratio correct non integer gba game scaling. :)

notimp, 2021-05-07:
Quick tip for RG351V owners, while trying to replicate the tip above (lcd3x filter) without RGA scaling being available on the latest builds, I came across the following scanline settings, that look decent.

Enable scanlines1280x720.cfg in the nesguy folder, and set overlay opacity to 0.30 (or 0.25 but 0.30 produces a darker image, which is beneficial to perceived black level), all under the On-Screen-Overlay setting.

Setting this as an overlay has the benefit of causing no performance penalty in game.

The goal still is to darken the overall image on screen, so you can use a one level higher brightness setting, to compensate for the horrible color accuracy of the screen on lower brightness settings.

If some of you alway notice a "beige" tint on screen - this combats it effectively. :) Of course you now have a scanline 'look' on your games, but even for people who dont like that, thats a small price to pay to get an overall better looking image out of the RG351V on lower brightness levels.

I've anyone knows where to get an overlay .cfg with scanlines for the native 640x480 resolution, I'd love to try that as well - but as is, retroarch only comes with scanline overlays for 1280x720 out of the box. There is no pixel distortion (apart from non integer scaling) but still, Id love to try this with a 640x480 scanline overlay as well.

Tested with SNES and Genesis games. Looks good. (Decent enough.) ;) From memory I'd say that lcd3x with RGA scaling lead to a more pleasant image but since RGA scaling is now gone as an option, this is a decent way to achieve a similar result in current builds. (Also the lcd3x filter had the downside, that if used with 480p (native) games, it sometimes would 'bug out' and only cause a very noticable green tint on screen. But for SNES or Genesis games, the combination with RGA scaling on, produced a nice result)

notimp, 2021-05-13:
N64 seemingly isn't being picked up, if only .zip files are in the n64 folder. Please fix, if you can reproduce it.

christianhaitian, 2021-05-13:
.zip are not supported by the mupen64plus standalone emulator which is the best emulator to use for N64 games.  That's why .zip does not show up.  Just use the supported extensions as shown in the link here: https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#nintendo-64

notimp, 2021-05-20:
You can stop implementing/maintaining retrorun32 - imho.. ;)

PSA: MUCH(!) better DC and GBA performance using the OpenAL audio core

Make sure to save the config file before exiting.

FFVI (MGBA) works with almost no audio stutter during dialogs
SoulCalibur (flycast-rumble, using retroarch32) runs much faster with almost no audio stutter (Fast MicroSD (20MB write), VGA, region ntsc, Render to texture: on)

Tested on ArkOS.

With the default alsathread audio core, there are frequent very noticeable audio stutters throughout both games, any much more slowdowns during SoulCalibur gameplay.

Imho try switching to OpenAL for a few other systems as well, and see what results it produces. :)
--

On a different note, the current 351ELEC is based on Retroarch 1.9.2 but still has RGA scaling as an option, and with it enabled (main menu), still manages 3:2 scaling (non-integer) for GBA games.

This makes the resulting image on GBA games (without filters) much more pleasing. ("Health Warning Screen" Font appears to not have any double lines on very fine grained (one pixel) font.)

But you know I'm sticking with ArkOS. ;)

notimp, 2021-05-21:
PPSSPP only outputs wrong aspect ratios. The image will be stretched on the y-achsis, no matter the setting.

See: https://i.imgur.com/Rj9U6Lo.jpeg

Please fix as soon as possible, this is most problematic high tier issue I've come across so far.

christianhaitian, 2021-05-21:
I couldn’t fix the stock one anyway as it came from the stock firmware.  My guess is it was stretched on stock as well.

notimp, 2021-05-29:
Small request, subjective. When starting up the RG351V the rumble pulse lasts for about 3 seconds and is quite abrasive. 351ELEC pulses only for a second or so and after the 50th time or so, you start to appreciate it, imho. Just something about how you feel, turning the system on over time. As it is entirely subjective its hard for me to argue for it - but subjectively this being less abrasive 'feels' better over time. :)

christianhaitian, 2021-05-29:
Have you flashed the stable image or just upgraded from the test release?  The stable image buzzes a little less longer.

notimp, 2021-06-03:
> Have you flashed the stable image or just upgraded from the test release? The stable image buzzes a little less longer.

Upgraded - thank you for the information. Will try. We just got another request for 'less annoing buzz please" on reddit (https://old.reddit.com/r/rg351v/comments/nqggqo/i_love_this_device_so_much_but_here_are_my/), not sure if the person is using ArkOS though.

Just saying.. ;)