christianhaitian/arkos #310: About Capcom (CPS-3) Emulation issues

vdincman, 2021-04-01:
There are casual frame drops, slow downs and stutters on CPS-3 games in general ( Street fighter 3rd Strike: Fight for the Future as example ). The best performance i could get so far is from Retroarch32 emulator with FBAlpha2016 core, but core options are rather limited considering other operating systems on this emulator. Is there a possibille performance gain with other Libretro Cores? If so is it possible to implement these cores under Capcom System? As far as i can see, the Emulation performance of Other OS for CPS-3 games are slightly superior than ArkOs at the moment. Thank you for your understanding.

christianhaitian, 2021-04-01:
I play that game regularly and I don’t suffer that issue using fbneo.  Are you using a good brand and authentic SD card such as Sandisk or Samsung?

christianhaitian, 2021-04-02:
As another follow up, using a FBAlpha v0.2.97.44 sf3 third strike rom, my matches are smooth and flawless.  I did notice when I had frame throttling on, audio was crackly.  Make sure you’re not using frame throttle settings such as rewind or fast forward.

vdincman, 2021-04-02:
I am using genuine Sandisk Ultra, Extreme and Samsung Evo Plus cards, i currently have 4 cards in total that is dedicated to my RG351P with 3 different OS built into them.
1- ArkOs - Sandisk Extreme 128GB
2- Arkos - Samsung Evo Plus 256GB
3- 351Elec - Sandisk Ultra 128GB
4- Anbernic Stock OS - NoName 64GB

For comparison, 351Elec and Anbernic Stock OS has no stutters or slowdowns while performing special moves, dashing and when there is a flashy scene on the game, on my both ArkOs builds there are from time to time. Even NoName Card with Stock OS doesn't have any issues running these games in particular, so i think the issue i am having isn't related with Card. I am 100% sure that my SD Cards are genuine, I always buy my cards from reputable stores, for example i bought my Samsung Evo Plus Card from an Official Samsung Store.
For the frame throttle, I've never used fast forward and rewind functions. I rarely play strategic RPG games, i double checked it, and they are off by default. 
I also tried to switch between the Hi clock And Normal Clock from the Advanced Menu section, but it didn't helped. 
I am a huge Street Fighter Fan and CPS-3 versions of it are my all time favorites. I love using ArkOs and it is my preferred OS by far, that's why I have 2 different builds from it, and i think what you have done with it is spectacular, i can't thank you enough for this. If you can help me resolve my problem with this particular issue, i will really appreciate it. I only switch to 351Elec for playing only a handful of CPS-3 games, and i really want to migrate to ArkOs completely.

christianhaitian, 2021-04-02:
I have no clue how to address an issue I can't replicate on my end.  I'm a huge SF fan myself and if I had that issue, I would definitely try to resolve it but I'm not.  Perhaps try to do core updates in retroarch to see if maybe some updates resolve it?

christianhaitian, 2021-04-02:
How did you flash the SD cards and have you updated your builds online to the latest 1.6 03182021-1 build?

vdincman, 2021-04-02:
> How did you flash the SD cards and have you updated your builds online to the latest 1.6 03182021-1 build?

Yes, both my builds are up to date. I am checking for new updates on a daily basis tbh. One of my builds is from V1,5 fresh ArkOS, and then updated to the latest version, and my other build is a ready to image called RetroArcade V.03 and they both have same performance. 
Can I ask which emulator and core are you using? I am using Retroarch32 as emulator and FBALPHA2016 as the core.

vdincman, 2021-04-02:
[image]
I've checked my 351Elec core settings and it is using "Arcade(FinalBurn Neo)" core, and it has smooth gameplay with no stutters or slowdowns even performing super special moves, parry, and dash.

christianhaitian, 2021-04-02:
The default emulator and core for cps3 in ArkOS is retroarch and fbneo.  Again, it’s smooth on my end.  What program did you use to image the sd card with ArkOS.

vdincman, 2021-04-02:
I am using a Macbook Pro and i used Balena Etcher to write the image to Sd card

christianhaitian, 2021-04-02:
That's your problem.  Balena Etcher is not recommended for imaging ArkOS per the instructions I provided on the main page.  It will cause strange performance issues like your experiencing now.  For Mac OS X, Apple Pi Baker is best.  Look over the Mac OS X instructions on the main page and make sure you're following that.

vdincman, 2021-04-02:
> The default emulator and core for cps3 in ArkOS is retroarch and fbneo. Again, it’s smooth on my end. What program did you use to image the sd card with ArkOS.

I know that it is like this by default. However the emulation by that configuration is actually like how RG350M emulates CPS-3 games, very choppy and stuttering. And every other system runs as is should be, so i don't see a reason for the program that i wrote can effect only that particular part of the image. I have witnessed other people having problems with CPS-3 games with regular Retroarch emulator, so i advised them to switch Retroarch32 and FBALPHA2016 Core to improve the performance. If you can DM me on FB, i can shoot a video and send it for you to see if you don't mind. My name is Vahit Dincman.

christianhaitian, 2021-04-02:
I am not on and don't use facebook.  I have checked with other users on the RGHandheld discord who have confirmed with me that the default retroarch and fbneo core give them smooth game play with sfiii 3rd strike on the rg351p/m, rgb10, oga, and rk2020.  They all confirmed as well that they following my imaging recommendation and have not experienced issues.  I can't explain why Balena Etcher causes issues with ArkOS with just writing, but I have witnessed that it does.  At this point, I will close this issue because there's not anything more I can do with this issue since I can't get anyone else to replicate this issue and you're using an unsupported image writing tool for this image.  I guess for CPS3, use 351elec or stock.  Take care.

closed 2021-04-02 by christianhaitian

vdincman, 2021-04-02:
Thanks for your time anyways, thumbs down was done by mistake by the way. I was trying to put a sad face haha 😅
I will try to write it again with a recommended software and try again.